open62541/open62541 #941: Some performance and quality questions

evminaev, 2017-01-21:
Questions
===========
1. If i add large number of nodes to server, than time depends not linear. For instance:
5000 - 0.4, 10000 - 0.9, 20000 - 2.2, 50000 - 12 seconds. I was too lasy to wait for 100000 nodes adding (typical number for plant servers).
Time mostly lost in getTypeHierarchy (as i anderstand). I guess that it is used for finding nodes with equal NodeID. May be such verification must be used optionaly? 

2. Function UA_Server_writeValue ( __UA_Server_write ) does not accept time and quality of value. Value can be from remote source and spend some time to reach server. How can be set time of aqusition? The same problem with quality. Is it wrong function, or time and quality realy does not suppored?

jpfr, 2017-01-24:
Hey @evminaev ,

1. Thanks for the hint. The increase is not so big (you loose a factor 3 per node between 5000 and 50000 nodes). But it should in fact be linear. It could be that lookups in the nodestore take longer as the hashmap fills up. We'll look into that.

2. You can use UA_Server_write, the "raw" write service, to write a value with statuscode and source timestamp. We may provide a convenient wrapper for this use case at a later point.

evminaev, 2017-03-20:
I use UA_Client_Service_browse to browse server structure from client. But it returned small number (5) of nodes only. Should i call it repeatedly, or i should call UA_Client_Service_browseNext? May be i use wrong method? The aim is to get list of nodes. Example in client.c is too simple and does not help.
Is there any other example?

jpfr, 2017-04-22:
`UA_Client_Service_browse` takes the maximum number of returned references as an argument.
Zero is the default and means `return everything`.
Have you made sure that more than 5 references correspond to the browse request?

jpfr, 2017-04-22:
Regarding the performance question.

> Time mostly lost in getTypeHierarch

I can confirm. Every variable instance is added to its variabletype with a 'hastypedefinition' reference.
When adding a new variable, we look for children of the variabletype to instantiate. Therefore, we loop over all references of the variabletype. This means we have a quadratic blow-up in terms of nodes of the same type.

One possibility is to move away from a simple array of references in the nodes to another representation.
For example a nested linked list. Then we can "jump over" references that are not interesting to us.

```
[0] backwards references 'hastypedefinition'
    - target nodeid 1
    - target nodeid 2
    - target nodeid 3
[1] forward references 'hascomponent'
   - target nodeid 4
   - target nodeid 5
   - target nodeid 6
```

Imho, changing the node layout should happen after 0.2.
But we need to address this.

@FlorianPalm @JGrothoff
We should have an understanding before changing the internal node representation.
As this would affect the nodestore plugins in #900 .

jpfr, 2017-05-31:
PR #1058 resolves the issue.
It still needs a bit more testing, therefore the PR is marked as WIP.

jpfr, 2017-06-05:
Solved via #1058.
Adding 100,000 nodes now takes milliseconds.

closed 2017-06-05 by jpfr

StalderT, 2017-07-23:
Do you plan to include PR #1058 in branch 0.2?

Thanks in advance?