open62541/open62541 #942: Use after free when Chunking

Andreas-N, 2017-01-23:
Hi,
I'm using your opc-ua sdk to implement a server, which not only provides single values, but also binary arrays of up to 65K Bytes. When using the UAExpert as a client, I sometimes see the following error: "Decoding failed! 0x80B00000" in the log.
```
13:01:16.945Z|6|7E48A700* [uastack] OpcUa_SecureStream_Read: end of buffer reached: 32888 requested, 32517 available in buffer; need to skip)!
13:01:16.945Z|6|7E48A700* [uastack] OpcUa_SecureStream_Read: Buffer skip to index 1 of 3!
13:01:16.945Z|6|7E48A700* [uastack] OpcUa_SecureStream_Read: end of buffer reached: 32888 requested, 32200 available in buffer; need to skip)!
13:01:16.945Z|6|7E48A700* [uastack] OpcUa_SecureStream_Read: Buffer skip to index 2 of 3!
13:01:16.945Z|6|7E48A700* [uastack] OpcUa_SecureStream_Skip: end of buffer reached: 13434288 requested, 8 available in buffer; need to skip)!
13:01:16.945Z|2|7E48A700* [uastack] OpcUa_Channel_ResponseAvailable: Decoding failed! (0x80B00000)
```
Using valgrind to examine the open62541 server, you can see a "use after free" error (Invalid write of size 2) when the length variable is encoded into the beginning of buffer (stored in the old_pos variable)  in ua_types_encoding_binary.c:802 when chunking did occur during encoding. In this case the memory block pointed to be old_pos is already freed in socket_write, after the block has been sent over the network or a network error occurred.

The same  "use after free" can be observed then applying the following patch to the server example application:
```
diff --git a/examples/server.c b/examples/server.c
index 19a2c9c..8fc6658 100644
--- a/examples/server.c
+++ b/examples/server.c
@@ -257,7 +257,7 @@ int main(int argc, char** argv) {
         UA_Variant_deleteMembers(&attr.value);
          /* add an array node for every built-in type */
-        UA_Variant_setArray(&attr.value, UA_Array_new(10, &UA_TYPES[type]), 10, &UA_TYPES[type]);
+        UA_Variant_setArray(&attr.value, UA_Array_new(32768, &UA_TYPES[type]), 32768, &UA_TYPES[type]);
         UA_Server_addVariableNode(server, UA_NODEID_NUMERIC(1, ++id), UA_NODEID_NUMERIC(1, ARRAYID),
                                   UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES), qualifiedName,
                                   UA_NODEID_NULL, attr, NULL, NULL);
```
It doesn't provoke the  "Decoding failed! 0x80B00000" on the client side, but illustrate the problem on the server side. The valgrind log is the following:
```
==29727== Invalid write of size 2
==29727==    at 0x4C2DDC3: memcpy@@GLIBC_2.14 (in /usr/lib64/valgrind/vgpreload_memcheck-amd64-linux.so)
==29727==    by 0x40782A: UInt32_encodeBinary (ua_types_encoding_binary.c:228)
==29727==    by 0x407860: Int32_encodeBinary (ua_types_encoding_binary.c:238)
==29727==    by 0x4089C6: ExtensionObject_encodeBinary (ua_types_encoding_binary.c:802)
==29727==    by 0x407C6B: Array_encodeBinary (ua_types_encoding_binary.c:476)
==29727==    by 0x409B5E: UA_encodeBinaryInternal (ua_types_encoding_binary.c:1221)
==29727==    by 0x409ADF: UA_encodeBinaryInternal (ua_types_encoding_binary.c:1207)
==29727==    by 0x409C0A: UA_encodeBinary (ua_types_encoding_binary.c:1244)
==29727==    by 0x40BCDA: UA_SecureChannel_sendBinaryMessage (ua_securechannel.c:226)
==29727==    by 0x4266AA: UA_Subscription_publishCallback (ua_subscription.c:525)
==29727==    by 0x416F12: processJob (ua_server_worker.c:64)
==29727==    by 0x417216: processRepeatedJobs (ua_server_worker.c:259)
==29727==  Address 0x74fd0d6 is 86 bytes inside a block of size 65,535 free'd
==29727==    at 0x4C2A3CC: free (in /usr/lib64/valgrind/vgpreload_memcheck-amd64-linux.so)
==29727==    by 0x40723B: UA_Array_delete (ua_types.c:1033)
==29727==    by 0x406F88: deleteMembers_noInit (ua_types.c:956)
==29727==    by 0x406FC5: UA_deleteMembers (ua_types.c:964)
==29727==    by 0x42DFC6: UA_ByteString_deleteMembers (ua_types_generated_handling.h:389)
==29727==    by 0x42E0DD: socket_write (ua_network_tcp.c:118)
==29727==    by 0x40BB01: UA_SecureChannel_sendChunk (ua_securechannel.c:174)
==29727==    by 0x407329: exchangeBuffer (ua_types_encoding_binary.c:77)
==29727==    by 0x407CAA: Array_encodeBinary (ua_types_encoding_binary.c:483)
==29727==    by 0x408E50: Variant_encodeBinary (ua_types_encoding_binary.c:908)
==29727==    by 0x409403: DataValue_encodeBinary (ua_types_encoding_binary.c:1036)
==29727==    by 0x409ADF: UA_encodeBinaryInternal (ua_types_encoding_binary.c:1207)
==29727==  Block was alloc'd at
==29727==    at 0x4C29160: malloc (in /usr/lib64/valgrind/vgpreload_memcheck-amd64-linux.so)
==29727==    by 0x40523B: UA_ByteString_allocBuffer (ua_types.c:182)
==29727==    by 0x42E345: ServerNetworkLayerGetSendBuffer (ua_network_tcp.c:276)
==29727==    by 0x40BBCA: UA_SecureChannel_sendBinaryMessage (ua_securechannel.c:198)
==29727==    by 0x4266AA: UA_Subscription_publishCallback (ua_subscription.c:525)
==29727==    by 0x416F12: processJob (ua_server_worker.c:64)
==29727==    by 0x417216: processRepeatedJobs (ua_server_worker.c:259)
==29727==    by 0x417848: UA_Server_run_iterate (ua_server_worker.c:600)
==29727==    by 0x417B2B: UA_Server_run (ua_server_worker.c:700)
==29727==    by 0x4049CB: main (server.c:397)
```
The problematic code lies in the function ExtensionObject_encodeBinary in ua_types_encoding_binary.c:793ff where the `pos` pointer is stored in the variable `old_pos` and is restored after encoding the payload, for writing the length to the beginning of the block. When chunking happens during the encoding process, because the maximum message length has been reached, the pointer in old_pos becomes invalid.

Another problematic code section is in my opinion in ua_types_encoding_binary.c:80, where encodeBuf is saved in a context variable and restored after exchangeBufferCallback has been called.
```
static UA_StatusCode
exchangeBuffer(void) {
    if(!exchangeBufferCallback)
        return UA_STATUSCODE_BADENCODINGERROR;

    /* store context variables since chunk-sending might call UA_encode itself */
    UA_ByteString *store_encodeBuf = encodeBuf;
    UA_exchangeEncodeBuffer store_exchangeBufferCallback = exchangeBufferCallback;
    void *store_exchangeBufferCallbackHandle = exchangeBufferCallbackHandle;

    size_t offset = ((uintptr_t)pos - (uintptr_t)encodeBuf->data) / sizeof(UA_Byte);
    UA_StatusCode retval = exchangeBufferCallback(exchangeBufferCallbackHandle, encodeBuf, offset);

    /* restore context variables */
    encodeBuf = store_encodeBuf;
    exchangeBufferCallback = store_exchangeBufferCallback;
    exchangeBufferCallbackHandle = store_exchangeBufferCallbackHandle;

    /* set pos and end in order to continue encoding */
    pos = encodeBuf->data;
    end = &encodeBuf->data[encodeBuf->length];
    return retval;
}
```
Restoring the pointer will end in a pointer to an already freed area, because the memory is freed in the socket_write function. 

Best regards 
Andreas

Background Information / Reproduction Steps
===========================================

Used Enviroment
 - [x ] open62541 Version (release number or git tag): branch 0.2; commit: cd3520db5e7b8f7614d9343c176f22fbf9b760b4
 - [x ] Other OPC UA SDKs used (client or server): UA Expert 1.4.3 269
 - [x ] Operating system: OpenSuse 42.2

jpfr, 2017-01-23:
Fixed via #943. I'll write up a post-mortem in a bit.

closed 2017-01-23 by jpfr

Andreas-N, 2017-01-24:
Thanks for the quick fix. Works like a charm. No more decoding failures in UAExpert.

jpfr, 2017-01-26:
> Another problematic code section is in my opinion in ua_types_encoding_binary.c:80, where encodeBuf is saved in a context variable and restored after exchangeBufferCallback has been called.

The encodeBuf is not restored. A pointer to the encodeBuf is restored.
The chunking procedure internally calls UA_encode to set the header of the chunk. So we need to be careful about restoring the current context.

I added more documentation (comments) to clarify the issue: 2f2facebaf2229a0d44cfcdcaedaa0666b9e93bf

commit e250d1b9 (2017-01-23): fix #942, encoding extensionobjects with chunking