open62541/open62541 #947: Add-Node in multithreading mode can lead to BadOutOfMemory-Error

janitza-luwe, 2017-01-24:
Description
===========
There is a Problem in multithread server when adding Nodes with _UA_Server_addDataSourceVariableNode()_. When doing this on a slower machine (ARM9 450Mhz in my case) with high scheduling rate, that function sometimes returns a "UA_STATUSCODE_BADOUTOFMEMORY"-Error. 

Background Information / Reproduction Steps
===========================================
Example code is unnecessary, imagine any code that creates a new Node with _UA_Server_addDataSourceVariableNode()_.

First this method creates a new node in Nodestore with _UA_NodeStore_newVariableNode()_ . After that the Node gets set up and subsequent methods are called with the **references** to this Node/Nodeid as parameter argument. One possible callstack is:

_UA_Server_addDataSourceVariableNode()_
--> _Service_AddNodes_existing()_
--> _instantiateNode()_
--> _Service_AddReferences_single()_
--> _UA_Server_editNode()_

This call to _UA_Server_editNode()_ in multithreading mode leads to a replacement of the Node in Nodestore by _UA_NodeStore_replace()_ --> **call_rcu(&oldEntry->rcu_head, deleteEntry)**

The deletion of the old Node in Nodestore is done by liburcu in a separate Thread (strace showed a syscall to 'clone()' which leads me to the problem). Since deletion is done without any locks, this happens as soon as the scheduler switches to the new created liburcu thread.
**The problem is, that the node-creating process continues working with the reference to the original created Node/NodeId (which is possibly deleted concurrently)!** 
In further execution another call to _UA_Server_editNode()_ will lead to a call to _UA_NodeStore_getCopy()_ with a NodeId-Reference to the (possibly) deleted Node, which is all 0 in that case. The lookup will fail to find the NodeId and  _UA_NodeStore_getCopy()_ returns NULL (which results in a UA_STATUSCODE_BADOUTOFMEMORY)

A solution could be to add a new Function _deleteEntry_rcu() that takes RCU-Lock before deleting the Node. This will preserve the referenced NodeID until RCU lock is released in _UA_Server_addDataSourceVariableNode()_.

Checklist
=========
 - [ ] open62541 Version (release number or git tag): Branch 0.2,  Commit-Id 21b8c18
 - [ ] Operating system: Linux 4.6 on ARM9 (ARMv5te)

jpfr, 2017-01-26:
Hello @janitza-luwe,

> The problem is, that the node-creating process continues working with the reference to the original created Node/NodeId (which is possibly deleted concurrently)!

That should not happen. Deleting nodes is delayed until all threads leave their current read-side ciritical section. I'll have a look at this. But I don't have time to deep-dive into the issue right now.

janitza-luwe, 2017-01-26:
As i mentioned, i did not understand locking at all. It does not seam that locking has any effect, as in debug build the lock is never unlocked:
```
# ifdef NDEBUG
...
# else
   extern UA_THREAD_LOCAL bool rcu_locked;
#   define UA_ASSERT_RCU_LOCKED() assert(rcu_locked)
#   define UA_ASSERT_RCU_UNLOCKED() assert(!rcu_locked)
#   define UA_RCU_LOCK() do {                     \
        UA_ASSERT_RCU_UNLOCKED();                 \
        rcu_locked = true;                        \
        rcu_read_lock(); } while(0)
#   define UA_RCU_UNLOCK() do {                   \
        UA_ASSERT_RCU_LOCKED();                   \
        rcu_locked = false;                       \
        rcu_read_lock(); } while(0)  <---- Lock again!?
# endif
```
But on my system the node gets erased concurrently! Is it no problem that no locks get unlocked but locked again on _UA_RCU_UNLOCK()_??

For now the patch is a good workaround for me. Moreover it should be preferred to work with call-by-value on access to nodestore, as done in most cases anyway.

jpfr, 2017-01-26:
Hello Lukas,

*ouch*. Good find.
I guess it was not enough to rely on valgrind saying "no leaks possible".
In effect, during debugging of multithreaded servers, memory was only freed during shutdown of the server. Fixed on 0.2 and master.

The "locks" only purpose is to test whether we nest read-side critical sections.
Which is allowed by the RCU mechanism but hints at a logic issue.

Can you elaborate why call-by-value is preferred?
It is quite convenient in the public API since we can create NodeIds in-situ.
But internally there is quite a bit of overhead involved.

janitza-luwe, 2017-01-26:
Sorry, i was talking about public API. It just must be impossible for the user-code to work with direct references to the nodestore. Of course, internal it should be no problem, if everything works as you described (i.e. _call_rcu()_-calls are _never_ executed while in RCU_LOCKED-section).
Unfortunately this does not seam to work in my case. Maybe its a bug in the liburcu i used? Iam on v0.9.2 (build for ARMv5te), maybe i should test v0.9.3...

jpfr, 2017-01-26:
> call_rcu()-calls are never executed while in RCU_LOCKED-section

Close but not quite right. Such a thing could only be achieved with blocking. RCU is alwas nonblocking.

call_rcu callbacks are executed when all threads have left their current critical section **if** they were in a critical section when the callback was created. But some threads might already be in **the next critical section** when the callback is actually run.

The idea is that nodes are always immutable if multithreading is enabled. In order to make changes, one has to make a copy. But when can the old node be deleted? We replace the node in the hashmap, so that no other thread can get a pointer to the old node. All operations on nodes happen inside a read-critical section. So when the call_rcu callback is executed, we know that no thread holds a pointer to the node (and cannot get one from the nodestore).

As a result, threads never block and are as fast as possible. The overhead for the callbacks is negligible.
The Linux kernel does the same, e.g. for concurrent file system operations. However, this ties us to a specific operating system. So plans are to move away from RCU in the long term but keep the public API stable. But as always, resources are scarce.

jpfr, 2017-01-26:
Btw, multithreading is still marked as *experimental*.
We need to have a master student or similar do a more formal verification.
But its already useful for us in non-mission critical projects.

jpfr, 2017-02-08:
I added a test that the function in question is only called inside a read-side critical section.

@janitza-luwe did you encounter the BadOutOfMemory on an actual system or was this hypothetical?
I double-checked and what you describe still seems odd.
Now the assertion should catch the case. Can you provide a stack-trace when you encounter the issue again?

closed 2017-02-08 by jpfr

janitza-luwe, 2017-02-08:
@jpfr it was an actual system, i debugged this very intensive. As i said above the BadOutOfMemory error rises out of an empty NodeId in _UA_NodeStore_getCopy()_. The point where the NodeID gets lost is not to be identified, as the Node gets erased concurrently. But it was always after _call_rcu(&oldEntry->rcu_head, deleteEntry)_ (see callstack above) and before leaving UA_RCU_LOCK-Section.

jpfr, 2017-02-08:
Hmm. Ok. Maybe I'll try to come up with a code snippet that reproduces the issue.

commit b0a23e7c (2017-02-08): add a test for addnode with multithreading enabled (fix #947)