open62541/open62541 #948: Calling functions that return arrays doesn't work (0.2rc)

samerafach, 2017-01-25:
Description
===========
Just taking the server_method.c and compiling it with g++ will not work properly on UAExpert.

Background Information / Reproduction Steps
===========================================
1. Get the following files and add them to your project:

https://github.com/open62541/open62541/releases/download/0.2-rc2/open62541.h
https://github.com/open62541/open62541/releases/download/0.2-rc2/open62541.c

2. Compile that example with g++ (you'll get a compile error, simply cast the `realloc` calls to `(UA_Byte*)`.

3. Run the server

4. Open UAExpert, add the address of the server and connect to it.

5. On the Address Space, right click on "1dArrayExample", and choose "Call".

6. Fill any values in the arrays, and call

7. Error appears "BadTypeMismatch".

I tried to follow up the error with the debugger, and I saw that Open62541 makes wrong assumptions about sizes, and ends up comparing size 5 with size 1.

 
Checklist
=========
Please provide the following information:

 - [0.2rc ] open62541 Version (release number or git tag):
 - [Server ] Other OPC UA SDKs used (client or server): 
 - [ Debian Jessie] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [X] Critical issue

jpfr, 2017-01-26:
Fixed in 0d54ac2b5f91df93fb31ad95eaa3d94f5bc08185.

closed 2017-01-26 by jpfr

jpfr, 2017-01-26:
By the way, the master branch now compiles with g++ per default.

samerafach, 2017-01-26:
@jpfr  Actually the example "server_method.c" doesn't compile in this version. It gives the error:

error: invalid conversion from ‘UA_StatusCode (*)(void*, UA_NodeId, size_t, const UA_Variant*, size_t, UA_Variant*) {aka unsigned int (*)(void*, UA_NodeId, long unsigned int, const UA_Variant*, long unsigned int, UA_Variant*)}’ to ‘UA_MethodCallback {aka unsigned int (*)(void*, const UA_NodeId*, const UA_NodeId*, void*, long unsigned int, const UA_Variant*, long unsigned int, UA_Variant*)}’ [-fpermissive]
                             1, &inputArguments1, 1, &outputArguments1, NULL);

Any quick tip that could help?