open62541/open62541 #951: decodeBinary returns non null terminated strings

Infinity95, 2017-01-27:
Description
===========
While trying to log a debug message with the bytestring returned by UA_AsymmetricAlgorithmSecurityHeader_decodeBinary(...) i noticed that the data is not null terminated. I believe this could be problematic? Or is this needed for some reason i don't know?
This might also be the case for other decode functions that return byte strings.

Background Information / Reproduction Steps
===========================================

Set breakpoint on line 272 in ua_server_binary.c, compile server_firstSteps and start it. Then connect to server with UaExpert client. On inspection of the data in asymHeaer.securityPolicyUri the string should not be null terminated.

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag): Branch master, dcc8248b0489c5ff6b4af43acce22c2b63f18b45
 - [ ] Other OPC UA SDKs used (client or server): Connecting to server with UaExpert
 - [ ] Operating system: Windows 10

jpfr, 2017-01-27:
Strings in OPC UA are not null terminated by length prefixed.
That way, we can also transport binary data where the string actually contains nulls.

Did this answer the question?

Infinity95, 2017-01-27:
Partly. I understand now that the strings should not be null terminated because they may conatin null and we know their length anyways.
Still, when logging a string the data needs to be passed and a null terminated string is required. Or is there another facility to log ByteStrings that i just have not stumbled across yet?

jpfr, 2017-01-27:
When logging a non-null terminated string, we set the length with the "%.*s" printf format.
When the string is known to be null-terminated (such as statically allocated strings as char *), "%s" is enough.

http://stackoverflow.com/questions/2239519/is-there-a-way-to-specify-how-many-characters-of-a-string-to-print-out-using-pri

Infinity95, 2017-01-27:
Thank you. I didn't know that. 👍

closed 2017-01-27 by Infinity95

jpfr, 2017-01-27:
You're welcome 😄