open62541/open62541 #952: server_method.c example doesn't compile on g++

samerafach, 2017-01-27:
Description
===========
the example "server_method.c" doesn't compile in this version. It gives the error:

error: invalid conversion from ‘UA_StatusCode ()(void, UA_NodeId, size_t, const UA_Variant*, size_t, UA_Variant*) {aka unsigned int ()(void, UA_NodeId, long unsigned int, const UA_Variant*, long unsigned int, UA_Variant*)}’ to ‘UA_MethodCallback {aka unsigned int ()(void, const UA_NodeId*, const UA_NodeId*, void*, long unsigned int, const UA_Variant*, long unsigned int, UA_Variant*)}’ [-fpermissive]
1, &inputArguments1, 1, &outputArguments1, NULL);

And other two similar errors for the other calls.

This error happens with the newest nightly release, on Debian Jessie (gcc 4.9.2).

You guys have a great library. You should make automated tests for everything.

Background Information / Reproduction Steps
===========================================

See here for the kinds of information we need to reproduce the bug and how to obtain it:
https://github.com/open62541/open62541/wiki/Writing-Good-Issue-Reports

Checklist
=========
Please provide the following information:

 - [X] open62541 Version (release number or git tag): Latest nightly (dcc8248)
 - [X] Other OPC UA SDKs used (client or server): Server
 - [X] Operating system: Debian Jessie
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [X] Self-contained code example attached
 - [X] Critical issue

jpfr, 2017-01-28:
Hmm. I cannot reproduce on the current master branch.
I tried with g++-6 and g++-4.8 on Debian Jessie. Both compile without an error.
What are your compiler flags?

samerafach, 2017-01-28:
Here's the compile line:

    g++ -c -pipe -g -std=gnu++11 -Wall -W -D_REENTRANT -fPIC -DQT_QML_DEBUG -o main.o ../main.cpp

It's generated automatically (I removed some -I includes for brevity). I'm using QtCreator for my project.

Are you sure you're referring to the same commit? What commit did you use? Could you give me the link of the compact cpp and h files you used?

jpfr, 2017-01-28:
Compiled with the current master branch.

```bash
jpfr@virtualbox:~/software/open62541/build$ g++-6 -pipe -g -std=gnu++11 -Wall -W -fPIC -Wno-write-strings -Wno-unused-parameter open62541.c server_method.c -o server
open62541.c:28354:37: warning: ‘visibility’ attribute ignored [-Wattributes]
 const UA_EXPORT UA_ConnectionConfig UA_ConnectionConfig_standard = {
                                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
open62541.c:28391:33: warning: ‘visibility’ attribute ignored [-Wattributes]
 const UA_EXPORT UA_ServerConfig UA_ServerConfig_standard = {
                                 ^~~~~~~~~~~~~~~~~~~~~~~~
open62541.c:28456:33: warning: ‘visibility’ attribute ignored [-Wattributes]
 const UA_EXPORT UA_ClientConfig UA_ClientConfig_standard = {
                                 ^~~~~~~~~~~~~~~~~~~~~~~~
```
```bash
jpfr@virtualbox:~/software/open62541/build$ g++-4.8 -pipe -g -std=gnu++11 -Wall -W -fPIC -Wno-write-strings -Wno-unused-parameter open62541.c server_method.c -o server
open62541.c:28354:37: warning: ‘visibility’ attribute ignored [-Wattributes]
 const UA_EXPORT UA_ConnectionConfig UA_ConnectionConfig_standard = {
                                     ^
open62541.c:28391:33: warning: ‘visibility’ attribute ignored [-Wattributes]
 const UA_EXPORT UA_ServerConfig UA_ServerConfig_standard = {
                                 ^
open62541.c:28456:33: warning: ‘visibility’ attribute ignored [-Wattributes]
 const UA_EXPORT UA_ClientConfig UA_ClientConfig_standard = {
```

I added `-Wno-write-strings -Wno-unused-parameter` to remove most of the warnings when compiling with C++.

samerafach, 2017-01-28:
OK. I got another nightly release and it worked. I'm really lost here. Thanks and sorry for wasting your time.

closed 2017-01-28 by samerafach

jpfr, 2017-01-28:
It's not wasting time. All issue reports are welcome so that we improve.
It was just chance that we caught it before.