open62541/open62541 #962: Compiling with OSX

suhamera, 2017-02-14:
I'm trying to compile the example from the main page on my Mac OS with gcc: gcc -std=c99 open62541.c server_variable.c -o server

As a result, I get the following error:
Undefined symbols for architecture x86_64:
  "_UA_ConnectionConfig_standard", referenced from:
      _main in server_variable-3fa5b6.o
  "_UA_Log_Stdout", referenced from:
      _logger in server_variable-3fa5b6.o
  "_UA_NODEID_NULL", referenced from:
      _main in server_variable-3fa5b6.o
  "_UA_ServerConfig_standard", referenced from:
      _main in server_variable-3fa5b6.o
  "_UA_ServerNetworkLayerTCP", referenced from:
      _main in server_variable-3fa5b6.o
  "_UA_Server_delete", referenced from:
      _main in server_variable-3fa5b6.o
  "_UA_Server_new", referenced from:
      _main in server_variable-3fa5b6.o
  "_UA_Server_run", referenced from:
      _main in server_variable-3fa5b6.o
  "_UA_Server_setVariableNode_valueCallback", referenced from:
      _main in server_variable-3fa5b6.o
  "_UA_TYPES", referenced from:
      _main in server_variable-3fa5b6.o
      _UA_Server_addVariableNode in server_variable-3fa5b6.o
  "_UA_Variant_setScalar", referenced from:
      _main in server_variable-3fa5b6.o
  "___UA_Server_addNode", referenced from:
      _UA_Server_addVariableNode in server_variable-3fa5b6.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)

Where do I need to specify the environment variables and where to find right values?

jpfr, 2017-02-15:
Hello @suhamera ,

it's funny that you use gcc but get a clang error.
Which version of open62541 are you using? Can you use -v to see the compiler invocation?

I can then reproduce and fix on my mac...

suhamera, 2017-02-15:
Hi, @jpfr thanx for your reply.

I've tried to compile for several platforms:
**OSX**
```
gcc -v
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
Apple LLVM version 8.0.0 (clang-800.0.42.1)
Target: x86_64-apple-darwin16.3.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin 

And i tried with Ubuntu and OpenWRT it was ok

Pro, 2017-03-08:
@suhamera what are you using as parameters for CMake?

I just tested building current 0.2 branch on OSX and it works fine. See #984 

`cmake -DCMAKE_BUILD_TYPE=Release -DUA_ENABLE_AMALGAMATION=ON -DUA_BUILD_EXAMPLES=ON ..`
and `make` should work right away.

See https://github.com/Pro/open62541/blob/eecf93ce4eb0f5b79f817ecbcb7d953b520525dc/tools/travis/travis_osx_script.sh#L32

jpfr, 2017-03-08:
Builds works on OSX and is tested on Travis.
Thanks @Pro!

Close via #984

closed 2017-03-08 by jpfr

maheshrd999, 2018-04-11:
I got the same error.
This is because my open62541.c file was empty some how.
Then I copy pasted the content and it started working fine now.

Hope it helps some one!