open62541/open62541 #963: connection to Tesla Scada

frax2222, 2017-02-16:
hello,

I have built an opc ua server from master branch for WEC7. I could connect to this opc ua server from UA expert and several other opc ua clients.
But when I connected from Tesla Scada I got connection error. I recorded tcp transfers with smartsniff (see attached files tesla.zip).
[tesla.zip](https://github.com/open62541/open62541/files/779403/tesla.zip)

Then I downloaded example server 64Bit, run it on Win7 I also got connection error (see tesla.jpg)
[image]

I contacted Tesla but they said that I should contact open62541.....

There was already a problem with Tesla last year (github.com/open62541/open62541/issues/689) but it seemed to be solved.

Has anybody else tried TeslaScada together with open62541 recently ?

Kind regards

jpfr, 2017-02-16:
The TeslaScada client is sending a CloseSecureChannel request as a normal message (MSG prefix).
They should send the CloseSecureChannel request with a CLO prefix.

[image]

@frax2222 Can you point the Tesla Scada developers to this issue? This seems to be a bug in their implementation, not ours.

jpfr, 2017-02-16:
By the way, Tesla Scada seems to be using the Java Stack from the OPC Foundation.
They fixed the issue in their Github repository recently.

https://github.com/OPCFoundation/UA-Java/issues/2

jpfr, 2017-02-20:
Fixed on the side of Tesla Scada.

closed 2017-02-20 by jpfr