open62541/open62541 #967: Can you help with generating bindings for Golang?

happierall, 2017-02-17:
I have some discussion [this](https://github.com/xlab/c-for-go/issues/20#issuecomment-280549371) about it.

1. include file not found: stdbool.h
2. open62541.h:1654:5: undefined: __builtin_object_size
3. open62541.h:1654:5: undefined: __builtin___memset_chk

Can you help? 
We can pay some money...

jpfr, 2017-02-20:
The issue was fixed in the c-for-go project.

closed 2017-02-20 by jpfr

gedw99, 2018-03-03:
@jpfr I know c for go reasonably well.
Can you point me to the code generation file that c for go uses ?

jpfr, 2018-03-03:
No idea. I closed this issue since it was reported as solved in the c-for-go project.

gedw99, 2018-03-03:
Thanks.

I found it in few minutes ago.
Will try to do a bringup and make a repo for it. Let me know if your
interested in adding it to your list of wrappers.

rtmaster, 2018-03-22:
hi @gedw99 can you kindly share how you were able to convert to Golang the library? Thanks!

gedw99, 2018-03-23:
I used this:

https://github.com/xlab/c-for-go

It's used for lots of similar requirements and the developer behind it is
very helpful.

I only did a quick test that it works.
I never had time to check the whole API.

I can probably help later.

gedw99, 2018-03-23:
> @gedw99 <https://github.com/gedw99> can you please share the YAML file
> you used in c-for-go?
> Also do you have an example of client in Go?
> Thanks a lot

Am really sorry but i dont have it anymore. I did a quikc 5 minute test and
then stopped when i saw it work.

The guy that is behind the cForGO code is really awesome and i am sure he
will help you if you let him knwo the code.
Best thing is to put up a git hub repo and let me and him knwo and we can
all work on it together.

Is that ok ?