open62541/open62541 #968: UA_Client_manuallyRenewSecureChannel() successfully, if connection is closed

Root-Core, 2017-02-19:
Description
===========

An call of `UA_Client_manuallyRenewSecureChannel()` returns `UA_STATUSCODE_GOOD` if the connection is closed.

This leads to errors in the `__UA_Client_Service()`, as the following `UA_SecureChannel_sendBinaryMessage()` call will fail with status `UA_STATUSCODE_BADCONNECTIONCLOSED`. This again will set the clients state to `UA_CLIENTSTATE_FAULTED`, but as the connection is dead it should be `UA_CLIENTSTATE_ERRORED`.

It would be set to `UA_CLIENTSTATE_ERRORED`, if the `UA_Client_manuallyRenewSecureChannel()` would return an faulted status.

Background Information / Reproduction Steps
===========================================

- Connect an client to an server
- Use eg. `UA_Client_Service_write()` to set values on the server
- Let the client loop
- If you now shutdown the server, the client will try to renew the secure channel in an loop 

I hope that's all, as the actual client is way more dynamic.
But this should be enough to reproduce the issue.

Checklist
=========

 - open62541 Version: 2.0 trunk (server and client)
 - Operating system: ELDK on ARM7
 - Critical issue: It might be, as the state machine is broken.

I added some additional logging to `__UA_Client_Service()` (just search "ROOT_CORE"):
https://gist.github.com/Root-Core/2e0924a175deca05297d7836824389f0

Output:
> ROOT_CORE: Renew success
> Sending a request of type 671
> ROOT_CORE: Binary message result: 80ae0000
> ROOT_CORE: Renew success
> Sending a request of type 671
> ROOT_CORE: Binary message result: 80ae0000
> [...]

Thank you all for your good work, keep it up.  :thumbsup:

jpfr, 2017-02-20:
We need to make this behaviour configurable for 0.3.
Maybe with boolean options in the client config
- autoReconnectSession
- autoReconnectSecureChannel

Then, an appropriate error is returned when autoreconnect is disabled *or* if autoreconnect fails.

For 0.2, returning the correct status code is fine.
@Root-Core Can you make a PR from the changes?
Concerning the logging: This is a good idea. In addition, you can use `UA_StatusCode_name` to get a "speaking" name of the statuscode instead of logging hex codes.

Best regards,
Julius

Root-Core, 2017-02-20:
I will look into it. Should I add this kind of logging there?
`UA_StatusCode_name` is a good idea, this logging here was just a fast hack. ;)

bostjanv, 2017-03-30:
Since it could be related to the same issue I'm commenting here.

I'm not sure what is the current expected behavior but I have identified two cases where closeSession callback is not being called (but it should have been in my opinion):
- When the session is removed due to timeout.
- When the client closes the connection unexpectedly.

And a somehow related question: Is there a way how to renew/reconnect closed session?

Best regards,
Bostjan

Pro, 2017-11-23:
@StalderT @Infinity95 this could be related to #1386 and #1336

StalderT, 2017-11-23:
Normally fix in https://github.com/open62541/open62541/commit/ffcfbc68f3798938dd04e4b969d018fc3ba88e01
https://github.com/open62541/open62541/blob/ffcfbc68f3798938dd04e4b969d018fc3ba88e01/src/client/ua_client_highlevel_subscriptions.c#L448-L449

StalderT, 2017-11-23:
@Pro normally https://github.com/open62541/open62541/pull/1386 https://github.com/open62541/open62541/pull/1336 and https://github.com/open62541/open62541/pull/1392 will fix many problems and provide a reconnect option. What do you want to add?

Pro, 2017-11-23:
@StalderT after your mentioned PR are merged, we should verify that the issue described by @Root-Core is really fixed, i.e. (see initial post):

> A call of UA_Client_manuallyRenewSecureChannel() returns UA_STATUSCODE_GOOD if the connection is closed.

StalderT, 2017-11-23:
OK. Let's wait and see.

```A call of UA_Client_manuallyRenewSecureChannel() returns UA_STATUSCODE_GOOD if the connection is closed.``` normally fix here : https://github.com/open62541/open62541/blob/ffcfbc68f3798938dd04e4b969d018fc3ba88e01/src/client/ua_client_highlevel_subscriptions.c#L448-L449

Pro, 2017-11-24:
@StalderT the PR is now merged. Could you test if this issue is fixed and close it if yes?
Thanks!

StalderT, 2017-11-25:
I will wait until https://github.com/open62541/open62541/pull/1397 will be merge and I will add a unit test

jpfr, 2017-12-06:
#1397 and the unit test are merged.

StalderT, 2017-12-07:
@Root-Core can you test and close the issue if resolved ? Thanks,

Root-Core, 2018-01-16:
Nice work, I will test it later this month as the priorities are shifting towards this project again.

closed 2018-02-01 by jpfr