open62541/open62541 #970: Random Null-Bytes in Amalgamation

Pro, 2017-02-20:
There seems to be a problem with the amalgamation in current 0.2 (and master) branch.

Travis and AppVeyor fail randomly at complete different positions.
E.g. travis fails with `error: unterminated #ifdef` in `open62541.c`
* https://travis-ci.org/open62541/open62541/jobs/203472064
* https://travis-ci.org/open62541/open62541/jobs/201922623#L3007
* https://travis-ci.org/Pro/open62541/jobs/203472199#L3000
* https://travis-ci.org/Pro/open62541/jobs/202260424#L2435

and AppVeyor with similar errors

* https://ci.appveyor.com/project/Stasik0/open62541/build/2479/job/9ytybu0b62bkwdq0#L371
* https://ci.appveyor.com/project/Stasik0/open62541/build/2477/job/v7aukrdkwvmv296g#L373)

These are exactly the same commits, but one failed, the other one was successful:
* Failed: https://ci.appveyor.com/project/Stasik0/open62541/build/2481
* Success: https://ci.appveyor.com/project/Pro/open62541/build/489

If you simply press the rebuild button on travis or AppVeyor, the build is successful this time (or after multiple tries).

This looks to me like a problem with the generation of NS0 files or amalgamation. I.e. there is a Null-Byte in the text file and thus the compiler just stops reading the file. 

Did anyone of you also experience this problem? Where could be the issue?

jpfr, 2017-02-20:
This looks like a bug on Travis' side.
The reason for that is simple: The amalgamation is deterministic (besides printing the date in the header).
So there is no reason whatsoever, why the procedure should works sometimes and fail at other times.

Pro, 2017-02-20:
Hmm, yes, that's also what I thought. So another reason could be that the same file system is shared between parallel builds, i.e., for Pull Requests there are two builds running at the same time. Maybe this causes file truncation.
Tried to create separate build directories for each build, but the error still persists.
https://travis-ci.org/Pro/open62541/jobs/203616487

Pro, 2017-02-20:
And the problem is on Travis AND Appveyor, I don't think that both CI systems have the same bug.

Could it be related to the NS0 compiler somehow?

I will see if I can do some additional tests/debugging

Pro, 2017-02-20:
Maybe flushing of the generated files helps. 

Here are some more outputs which should all succeed, but fail at different random lines:

https://travis-ci.org/Pro/open62541/jobs/203627541#L2741
https://travis-ci.org/open62541/open62541/jobs/203627464#L2473
https://travis-ci.org/open62541/open62541/jobs/203627473#L2481

https://ci.appveyor.com/project/Stasik0/open62541/build/2485/job/v8xgb11msagoae0q#L381

jpfr, 2017-02-23:
/tools/amalgamate.py is only a few lines short and looks ok.

We print the size of the generated files to the terminal. Are they sizes consistent between travis jobs?
Otherwise, we can print checksums with the `md5sum` command and compare travis / local generated files.

Pro, 2017-02-23:
I already found a way to upload the generated files from Travis right after amalgamation.

See https://github.com/Pro/open62541/tree/hotfix/null_byte?files=1
So we can download the file if it fails.

There it never happened, so it may be a timing issue maybe... I will investigate that further

Infinity95, 2017-02-27:
I'm currently running into problems with the amalgamation process as well. Im working on implementing security policies and the new headers i need for that all work well without amalgamation. The amalgamation process seems to be putting them in the generated header file in the wrong order, which in turn gives undefined identifier errors. Am i doing something wrong or might this be related?

As far as i saw, the amalgamation script just takes the files as argument and then concatenates them, while leaving out e.g. include directives.

Reordering the files in the CmakeLists.txt solves the problem, but i don't think this is desireable behavior.

Pro, 2017-02-27:
You can try to cherry pick my commit from the branch mentioned above, so you will be able to download the files which Travis/Appveyor creates.
Unfortunately I could not yet reproduce the problem....

jpfr, 2017-02-27:
> Reordering the files in the CmakeLists.txt solves the problem, but i don't think this is desireable behavior.

You **need** to have a correct ordering in the CMakeLists.txt. There's no other way.
And I don't see a problem with that either.

Pro, 2017-03-08:
I think I found the problem.
In summary:
When calling `file.close()`, it calls `file.flush()` which writes the content for the file from the program buffer to the OS buffer. So the data is not yet physically on the HDD. To force flushing the OS buffer one needs to call `os.fsync(file)`.
See: https://stackoverflow.com/questions/7127075/what-exactly-the-pythons-file-flush-is-doing

And a similar problem:
https://stackoverflow.com/questions/13761961/large-file-not-flushed-to-disk-immediately-after-calling-close

closed 2017-03-08 by Pro

jpfr, 2017-03-08:
Perfect!

commit b9526c01 (2017-02-20): Make sure build directories are distinct. Fix #970
commit 6e7e1253 (2017-03-08): Fix random build fails on travis/appveyor. Fix #970 

flush and fsync amalgamated file
commit 800bd617 (2017-03-08): flush and fsync amalgamated file. Fix #970