open62541/open62541 #972: LDS-ME server

symgyyb, 2017-02-21:
Description
===========

       When I call a FindServersOnNetwor service, can only get to host hostname, cannot but only know the IP address, host name and port number can not connect to the server, if you want to get IP address, what should I do?
      Hope to get your help, thank you very much.

Background Information / Reproduction Steps
===========================================

See here for the kinds of information we need to reproduce the bug and how to obtain it:
https://github.com/open62541/open62541/wiki/Writing-Good-Issue-Reports

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Pro, 2017-02-21:
Please fill out the checklist, otherwise it's hard to tell where the problem may be.

Did you use the `feature_mdns` branch?
Did you compile with `UA_ENABLE_DISCOVERY_MULTICAST=ON`? Otherwise FindServersOnNetwork will return `BadServiceNotSupported`

If I use the most recent feature_mdns version and the examples (https://github.com/open62541/open62541/tree/feature_mdns/examples/discovery) everything works fine.
* Start `example_discovery_server_lds`
* Start `example_discovery_server_register`
* Query the server with `example_discovery_client_find_servers`

It gives the following output, whereas the first two Server outputs are from FindServersOnNetwork:

```
Server[0]: LDS-hostname
	RecordID: 0
	Discovery URL: opc.tcp://hostname:4840
	Capabilities: LDS,

Server[1]: Sample Server-hostname
	RecordID: 1
	Discovery URL: opc.tcp://hostname:16664
	Capabilities: NA,

Server[0]: urn:open62541.example.local_discovery_server
	Name: open62541-based OPC UA Application
	Application URI: urn:open62541.example.local_discovery_server
	Product URI: http://open62541.org
	Type: Server
	Discovery URLs:
		[0]: opc.tcp://hostname:4840

Server[1]: urn:open62541.example.server_register
	Name: open62541-based OPC UA Application
	Application URI: urn:open62541.example.server_register
	Product URI: http://open62541.org
	Type: Server
	Discovery URLs:
		[0]: opc.tcp://hostname:16664

-------- Server Endpoints --------

Endpoints for Server[0]: urn:open62541.example.local_discovery_server

	Endpoint[0]:
		Endpoint URL: opc.tcp://hostname:4840
		Transport profile URI: http://opcfoundation.org/UA-Profile/Transport/uatcp-uasc-uabinary
		Security Mode: NoneSignSign and Encrypt
		Security profile URI: http://opcfoundation.org/UA/SecurityPolicy#None
		Security Level: 0
Endpoints for Server[1]: urn:open62541.example.server_register

	Endpoint[0]:
		Endpoint URL: opc.tcp://hostname:16664
		Transport profile URI: http://opcfoundation.org/UA-Profile/Transport/uatcp-uasc-uabinary
		Security Mode: NoneSignSign and Encrypt
		Security profile URI: http://opcfoundation.org/UA/SecurityPolicy#None
		Security Level: 0

```

Wtango, 2017-02-21:
@symgyyb  As a workaround, If you want your Open62541 OPCUA server reply the IP address instead of hostname when responding the Discovery service set, you can modify the code here:[ua_network_tcp.c](https://github.com/open62541/open62541/blob/0.2/plugins/ua_network_tcp.c#L380), change the hostname to an IP address.

Pro, 2017-02-21:
@symgyyb Ah, sorry, now I got what you asked.

According to the OPC UA Specification the endpoint should always contain the hostname, and not the IP address.
If you want to map the hostname to a specific IP, I strongly recommend using `/etc/hosts` (Linux) or ` C:\Windows\System32\drivers\etc\hosts` (Windows).
This way you can avoid changing the source code.

Pro, 2017-03-07:
Closing for now. Please reopen if you still have questions.

closed 2017-03-07 by Pro