open62541/open62541 #976: I Find a little Bug(Maybe) about the   accessLevel of Attributes

seahair, 2017-02-24:
Description
===========
I use the open62541_V2 ,and the union code(only two file: open62541.h and open62541.c), I find when I  use UA_Server_addDataSourceVariableNode() to  add node "current time", the accessLevel of Attributes  not display, even I  write:
v_attr.accessLevel = UA_ACCESSLEVELMASK_READ | UA_ACCESSLEVELMASK_WRITE;
the Uaexpert display none in  accessLevel.

but  when I  use  the open62541_V2  not the only  two files,  there is many many files, the problem is gone, so  I think this is a  little bug.
 
sorry, My English is very poor,so ~.~

Background Information / Reproduction Steps
===========================================

See here for the kinds of information we need to reproduce the bug and how to obtain it:
https://github.com/open62541/open62541/wiki/Writing-Good-Issue-Reports

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag): 0.2
 - [ ] Other OPC UA SDKs used (client or server): client : Uaexpert
 - [ ] Operating system: Ubuntu 16.04 LTS   64bit
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Stasik0, 2017-02-24:
Did you try one of the latest amalgamsted files from here:  http://open62541.org/releases/?

For example latest 0.2 branch? http://open62541.org/releases/776c27ba8b.zip

Its probably been fixed, but not yet released officially.

seahair, 2017-02-24:
ＯＫ　I have a  test, U are right, I use the released fixed code ,and the problem is gone, think you. 　上海　中国　欢迎你

closed 2017-02-24 by jpfr