open62541/open62541 #979: segmentation fault in processRepeatedJobs.

Russ2020, 2017-02-28:
===========
segmentation fault experienced in processRepeatedJobs when accessing member of server repeated jobs list.

Background Information / Reproduction Steps
===========================================
0.2.0-rc2 amalgamated source.

Updated example server_datasource.c to add two objects with a variable under them. Variable in incremented on read callback. Connected UAExpert client and subscribed to server status and two user variables. Ran server for a more than 24hrs.

/* This work is licensed under a Creative Commons CCZero 1.0 Universal License.
 * See http://creativecommons.org/publicdomain/zero/1.0/ for more information. */

#include <signal.h>
#include <stdio.h>

#ifdef UA_NO_AMALGAMATION
# include "ua_types.h"
# include "ua_server.h"
# include "ua_config_standard.h"
# include "ua_network_tcp.h"
# include "ua_log_stdout.h"
#else
# include "open62541.h"
#endif

UA_Boolean running = true;
UA_Logger logger = UA_Log_Stdout;

UA_Int32 myInteger = 42;

static void stopHandler(int sign) {
    UA_LOG_INFO(logger, UA_LOGCATEGORY_SERVER, "received ctrl-c");
    running = false;
}

static UA_StatusCode
readInteger(void *handle, const UA_NodeId nodeid, UA_Boolean sourceTimeStamp,
            const UA_NumericRange *range, UA_DataValue *dataValue) {
    dataValue->hasValue = true;
	
	myInteger++;
	
    UA_Variant_setScalarCopy(&dataValue->value, (UA_UInt32*)handle, &UA_TYPES[UA_TYPES_INT32]);
    // we know the nodeid is a string
    UA_LOG_INFO(logger, UA_LOGCATEGORY_USERLAND, "Node read %.*s",
                nodeid.identifier.string.length, nodeid.identifier.string.data);
    UA_LOG_INFO(logger, UA_LOGCATEGORY_USERLAND, "read value %i", *(UA_UInt32*)handle);
    return UA_STATUSCODE_GOOD;
}

static UA_StatusCode
writeInteger(void *handle, const UA_NodeId nodeid,
             const UA_Variant *data, const UA_NumericRange *range) {
    if(UA_Variant_isScalar(data) && data->type == &UA_TYPES[UA_TYPES_INT32] && data->data){
        *(UA_UInt32*)handle = *(UA_UInt32*)data->data;
    }
    // we know the nodeid is a string
    UA_LOG_INFO(logger, UA_LOGCATEGORY_USERLAND, "Node written %.*s",
                nodeid.identifier.string.length, nodeid.identifier.string.data);
    UA_LOG_INFO(logger, UA_LOGCATEGORY_USERLAND, "written value %i", *(UA_UInt32*)handle);
    return UA_STATUSCODE_GOOD;
}

int main(int argc, char** argv) {
    signal(SIGINT, stopHandler); /* catches ctrl-c */

    UA_ServerConfig config = UA_ServerConfig_standard;
    UA_ServerNetworkLayer nl = UA_ServerNetworkLayerTCP(UA_ConnectionConfig_standard, 16664);
    config.networkLayers = &nl;
    config.networkLayersSize = 1;
    UA_Server *server = UA_Server_new(config);
	
	UA_NodeId myObjectNodeId = UA_NODEID_NUMERIC(1, 1);
	UA_QualifiedName myObjectName = UA_QUALIFIEDNAME(1, "pump 1");
    UA_ObjectAttributes objattr;
    UA_ObjectAttributes_init(&objattr);
    objattr.description = UA_LOCALIZEDTEXT("en_US","pump 1");
    objattr.displayName = UA_LOCALIZEDTEXT("en_US","pump 1");
	UA_Server_addObjectNode(server, myObjectNodeId,
								UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
								UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES),
								myObjectName, UA_NODEID_NULL, objattr, NULL, NULL);
								
								

    /* add a variable node to the address space */
    //UA_Int32 myInteger = 42;
    UA_NodeId myIntegerNodeId = UA_NODEID_NUMERIC(1, 101);
    UA_QualifiedName myIntegerName = UA_QUALIFIEDNAME(1, "the answer");
    UA_DataSource dateDataSource = (UA_DataSource) {
        .handle = &myInteger, .read = readInteger, .write = writeInteger};
    UA_VariableAttributes attr;
    UA_VariableAttributes_init(&attr);
    attr.description = UA_LOCALIZEDTEXT("en_US","the answer");
    attr.displayName = UA_LOCALIZEDTEXT("en_US","the answer");

    UA_Server_addDataSourceVariableNode(server, myIntegerNodeId,
                                        myObjectNodeId,
										//UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
                                        UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES),
										myIntegerName, UA_NODEID_NULL, attr, dateDataSource, NULL);
										
	UA_NodeId myObjectNodeId2 = UA_NODEID_NUMERIC(1, 2);
	UA_QualifiedName myObjectName2 = UA_QUALIFIEDNAME(1, "pump 2");
    UA_ObjectAttributes objattr2;
    UA_ObjectAttributes_init(&objattr2);
    objattr2.description = UA_LOCALIZEDTEXT("en_US","pump 2");
    objattr2.displayName = UA_LOCALIZEDTEXT("en_US","pump 2");
	UA_Server_addObjectNode(server, myObjectNodeId2,
								UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
								UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES),
								myObjectName2, UA_NODEID_NULL, objattr2, NULL, NULL);
								
								

    /* add a variable node to the address space */
    //UA_Int32 myInteger = 42;
	// node id must be unique in namespace.
	UA_NodeId myIntegerNodeId2 = UA_NODEID_NUMERIC(1, 102);
    // Browse name.
	UA_QualifiedName myIntegerName2 = UA_QUALIFIEDNAME(1, "state");
    UA_DataSource dateDataSource2 = (UA_DataSource) {
        .handle = &myInteger, .read = readInteger, .write = writeInteger};
    UA_VariableAttributes attr2;
    UA_VariableAttributes_init(&attr2);
    attr2.description = UA_LOCALIZEDTEXT("en_US","the status of the pump");
    attr2.displayName = UA_LOCALIZEDTEXT("en_US","state");

    UA_Server_addDataSourceVariableNode(server, myIntegerNodeId2,
                                        myObjectNodeId2,
										//UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
                                        UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES),
										myIntegerName2, UA_NODEID_NULL, attr2, dateDataSource2, NULL);

    UA_StatusCode retval = UA_Server_run(server, &running);
    UA_Server_delete(server);
    nl.deleteMembers(&nl);

    return (int)retval;
}
[image]

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2017-02-28:
Hello @Russ2020 ,

I see a crash in this line:
```c
    UA_Variant_setScalarCopy(&dataValue->value, (UA_UInt32*)handle, &UA_TYPES[UA_TYPES_INT32]);
    // we know the nodeid is a string
   UA_LOG_INFO(logger, UA_LOGCATEGORY_USERLAND, "Node read %.*s",
                nodeid.identifier.string.length, nodeid.identifier.string.data);
```
The nodeid contains a numeric identifier. So the comment does not hold.

Can you resolve this issue first? I'm not able to debug far enough into the program to see the error you described above.

Best,
Julius

jpfr, 2017-03-20:
Hello @Russ2020 ,

please reopen if the last comment did not resolve your issue.

closed 2017-03-20 by jpfr