open62541/open62541 #983: I have succesfully used the open62541 library to connec to and interact with a OPC US Server.  However, can anyone provide a sample of how to use the open62541 library to enumerate OPC UA Servers on the local (or remote) PC?

mogulza, 2017-03-08:
Description
===========

Background Information / Reproduction Steps
===========================================

See here for the kinds of information we need to reproduce the bug and how to obtain it:
https://github.com/open62541/open62541/wiki/Writing-Good-Issue-Reports

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Pro, 2017-03-08:
Thanks for your question.
Next time please try to fill out the corresponding fields in the description area to provide enough information.

You can use the `FindServers` method call. An example is shown here:
https://github.com/open62541/open62541/blob/master/examples/discovery/client_find_servers.c
The corresponding function can be found here:
https://github.com/open62541/open62541/blob/master/src/client/ua_client.c#L660

Please note that for this functionality you probably need to use the `master` branch, instead of `0.2`.

Please close this issue if it answers your question.

mogulza, 2017-03-08:
Hi Pro ... thanks for the response.  

I am a first time user on this board so I was not sure exactly which fields to check/fill in relating to my question. Can you advise me on this please?

I just tried check checking the "open62541 Version" and "Operating System" check boxes but I see nowhere that I need to fill in the version or the operating system.  Or do I simply make that information part of my text query?

Pro, 2017-03-08:
No problem, happy to help:
You have to edit your post, and then you can add the needed information after the colon for each item.
Then check the boxes if the requested information is filled out.

You don't need to do that for this Issue if it already solves your question. Just close it :+1:

mogulza, 2017-03-08:
Downloaded master branch ... but my app now fails to compile as it cannot find the files "ua_types_generated.h" and "ua_types_generated_handling.h"

Do I need something else?

jpfr, 2017-03-08:
You can either follow the build instructions: http://open62541.org/doc/current/building.html
Or download a nightly build of the single-file release of the master branch where everything is included in a pair of .c and .h files: http://open62541.org/releases/

Please be aware that for servers to be "discoverable", they have to either
- register at a discovery server
- support multicast DNS
Registering at a discovery server is more widely supported at this time. Multicast DNS is nicer technology-wise and enables plug&work scenarios.

mogulza, 2017-03-08:
Thanks jpfr ... downloaded nightly build ... now compiling but UA_Client_findServersOnNetwork() returns 0x80020000.   Investigating ....

mogulza, 2017-03-08:
Not sure why I am getting the 0x80020000 error from UA_Client_findServersOnNetwork().

The Discovery Service is running and a 3rd party OPC UA client app is certainly able to enumerate the local OPC UA Servers.

Debugging down into the code, I find that _UA_Client_connect() due to EndpointsHandshake() failing because it is unable to find an endpoint with a securityPolicyUri, = securityNone !

Any advice?

Pro, 2017-03-08:
Which server are you using as the Discovery Server?

Due to your comment about securityPolicy, I suspect that you are using the OPC Foundation LDS server?
Since open62541 does not yet support encryption/security, you need to set up an endpoint without security.

This will allow you to at least call the FindServers and FindServersOnNetwork method.

RegisterServer and RegisterServer2 will not be possible with the OPC Foundation LDS, since this requires authentication and for authentication encryption is needed.

mogulza, 2017-03-08:
Thanks Pro ... 
I was  not aware that "open62541 does not yet support encryption/security".   Just found that in the PDF ! 

I'm afraid my investigation of using this library stops here.  Any idea on the timing for supporting encryption/security?

jpfr, 2017-03-08:
> Any idea on the timing for supporting encryption/security?

It will take a few more months. With emphasis on few.
We have a running messy prototype that works. Now we refactor the code into something maintainable.

mogulza, 2017-03-08:
Thanks all!

closed 2017-03-08 by mogulza