open62541/open62541 #987: UA client can not link to UA server

ruikai, 2017-03-13:
Description
===========
i use the order gcc -std=c99 my_server.c open62541.c -o test ;
and it generate test;
then  ./test;
and it is display the server is run ,and the Port number is 4840;
then i use the UAexpert to connect with it;
it shows could not connect to server:BadTimeout;

especially, i run it in the cortex-a8(linux) it is run ok;
but i run it in the cortex-a5,it shows the problem :Badtimeout

Background Information / Reproduction Steps
===========================================

See here for the kinds of information we need to reproduce the bug and how to obtain it:
https://github.com/open62541/open62541/wiki/Writing-Good-Issue-Reports

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2017-03-13:
Hello @ruikai ,

can you enable logging and post the output? See the wiki page on writing good issue reports for details.

Furthermore, are you able to handshake with the TCP port at all? For example via telnet...

ruikai, 2017-03-14:
[image]
Above is running server in cortex-a5,and it shows ok;
and then i use the UAexpert to connect with it;
and the output is below:
[image]
i do not know what is wrong whit it,and i run the server in PC and cortex-a8 is OK,and UAexpert connect is ok；
thank you;

jpfr, 2017-03-15:
Hello @ruikai ,

I traced the steps in the code.
The error occurs because the `validTill` timestamp of the session is in the past.
That's funny, because the timestamp is set to `UA_DateTime_nowMonotonic() + session->timeout` when the session is created (right before the activation).

Can you have a look at your implementation of UA_DateTime_nowMonotonic?
Just add some additional log information to the important places.

ruikai, 2017-03-16:
hello  @jpfr :

         thank you for you help,that is the problem,in the function UA_DateTime_nowMonotonic,there is a clock_gettime,when i Compile the project，i shoule add -lrt, and now ,it is works OK,Thanks very much；

ruikai, 2017-03-16:
hello @jpfr:
          
           another question, i use the server_variable.c open62541.c and open62541.h,and compile them with       
gcc -std=c99 -lrt server_variable.c open62541.c -o test,and now it is run OK,and  i use UAexpert to connect with it,and the picture is like below:
[image]
it is ok,but the value always be 42,my question is I want to see dynamic data changes，how can i do with it,thanks

jpfr, 2017-03-20:
Hello @ruikai ,

you already see dynamic data changes.
There is a variable /Objects/Server/ServerStatus/CurrentTime. If you add that to the data access view, you should see the changes.

Otherwise, you might have subscriptions disabled in the build settings.

closed 2017-03-20 by jpfr