open62541/open62541 #996: Large Array and outlier

jmrahm, 2017-03-17:
Description
===========
Hey guys,
for a project we used your OPC UA Stack. During some test we get advertent to some outlier with subscriptions and large arrays. Large means for example 7 array from 1000 up to 7000 values. Every array are filled in a separate thread with random values (10 up to 50). These arrays were subscribe and if we recording the packages with wireshark in every package is one and only one outlier.

Background Information / Reproduction Steps
===========================================
To reproduce this issue in a more simpler way as out whole project, we used your example "tutorial_server_variable.c" and added some arrays in the "addVariable()" function. A thread which updates the values sequentially are added in main():
`thread* updater = new std::thread(writeVariable, server);`

After we start our little program we connect to the server with UA Expert and add all arrays to the subscription field (100ms), all looks great. But if we recording parallel the package with wireshark (see below) there are some outliers.

Whats happened? I prepare a table which shows some configuration of the arrays and the index of the outlier:

|              | order |      |      |      |      |      |      | index Summe |
|--------------|-------|------|------|------|------|------|------|-------|
| array size   | 7000  | 6000 | 5000 | 4000 | 3000 | 2000 | 1000 | 28000 |
| outlir index |       |      |      |      | 2550 |      |      | 24450 |
|              | 7000  | 6000 | 5000 | 4000 | 3000 | 2000 | 1000 | 28000 |
| outlir index |       |      |      |      | 2549 |      |      | 24451 |
|              | 7000  | 6000 | 5000 | 4000 | 2000 | 1000 | 3000 | 28000 |
| outlir index |       |      |      |      |      | 551  |      | 24449 |
|              | 3000  | 1000 | 2000 | 4000 | 5000 | 6000 | -    | 21000 |
| outlir index |       |      |      |      |      | 1358 |      | 16358 |
|              | 3000  | 1000 | 2000 | 4000 | 5000 | 6000 | 7000 | 28000 |
| outlir index |       |      |      |      |      |      | 3543 | 24543 |
|              | 3000  | 1000 | 2000 | 4000 | 5000 | -    | -    | 15000 |
| outlir index |       |      |      |      | 2176 |      |      | 12176 |
|              | 3000  | 1000 | 2000 | 4000 | -    | -    | -    | 10000 |
| outlir index |       |      |      | 2176 |      |      |      | 8176  |

It doesn't matter in which order the array are subscripted. Furthermore a colleague of mine try the same  configuration with a LabView client and the same outlier occure. Hence we suggest the error occure in your stack.

Here some wireshark screenshots

[image]
[image]
[image]
[image]

Checklist
=========

 - [x] open62541 Version (release number or git tag): 0.2
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Ubuntu 16.10 and Ubuntu 16.04
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [x] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [ ] Critical issue

Code example:
`		

		static void
		addVariableArray(UA_Server *server, UA_NodeId nodeId, const int32_t sizeArray) {
    		UA_VariableAttributes attr;
    		UA_VariableAttributes_init(&attr);
		UA_Double vectorizedValue[sizeArray];
		UA_Variant_setArrayCopy(&attr.value, &vectorizedValue, sizeArray, &UA_TYPES[UA_TYPES_DOUBLE]);
		UA_NumericRange arrayRange; 
		arrayRange.dimensionsSize = 1; 
		UA_NumericRangeDimension scalarThisDimension = (UA_NumericRangeDimension){.min = 0, .max = (unsigned)sizeArray};
		arrayRange.dimensions = &scalarThisDimension; 
		UA_Variant_setRangeCopy(&attr.value, &vectorizedValue, sizeArray, arrayRange); 

    //UA_Variant_setScalar(&attr.value, &myInteger, &UA_TYPES[UA_TYPES_INT32]);
		string name = "the.answer_" + to_string(sizeArray);
    attr.description = UA_LOCALIZEDTEXT("en_US", (char*) name.c_str());
    attr.displayName = UA_LOCALIZEDTEXT("en_US", (char*) name.c_str());
    attr.dataType = UA_TYPES[UA_TYPES_DOUBLE].typeId;
    attr.valueRank = 1; /* array with one dimension */

    UA_QualifiedName myIntegerName = UA_QUALIFIEDNAME(1, "the answer");
    UA_NodeId parentNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER);
    UA_NodeId parentReferenceNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES);
    UA_Server_addVariableNode(server, nodeId, parentNodeId,
                              parentReferenceNodeId, myIntegerName,
                             UA_NODEID_NULL, attr, NULL, NULL);
		}

		static void
		writeVariable(UA_Server *server) {	
	
	while(true) {
    		UA_NodeId myIntegerNodeId = UA_NODEID_NULL; 

		for(int32_t i=1000; i < 65535; i=i+1000) {
			string name = "the.answer_" + to_string(i);
			myIntegerNodeId = UA_NODEID_STRING(1, (char*) name.c_str());
			UA_Double vectorizedValue[i];
			for(int32_t k = 0; k < i; k++) {
				if(k % 2 == 0) {
					vectorizedValue[k] = rand() % 6 + 1;
				}
				else {
					vectorizedValue[k] = rand() % 50 + 10;
				}
			}
		
				UA_Variant myVar;
				UA_Variant_init(&myVar);
				UA_Variant_setArray(&myVar, &vectorizedValue, i, &UA_TYPES[UA_TYPES_DOUBLE]);
				UA_Server_writeValue(server, myIntegerNodeId, myVar);
				
				/* 
				* Set the status code of the value to an error code. The function
				* UA_Server_write provides access to the raw service. The above
				* UA_Server_writeValue is syntactic sugar for writing a specific node
				* attribute with the write service. */
				UA_WriteValue wv;
				UA_WriteValue_init(&wv);
				wv.nodeId = myIntegerNodeId;
				wv.attributeId = UA_ATTRIBUTEID_VALUE;
				wv.value.status = UA_STATUSCODE_BADNOTCONNECTED;
				wv.value.hasStatus = true;
				UA_Server_write(server, &wv);

				/* Reset the variable to a good statuscode with a value */
				wv.value.hasStatus = false;
				wv.value.value = myVar;
				wv.value.hasValue = true;
				UA_Server_write(server, &wv);
				}
			}
		}

		UA_Boolean running = true;
		static void stopHandler(int sign) {
   		 UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "received ctrl-c");
   		 running = false;
		}

		int main(void) {
		UA_ServerConfig config = UA_ServerConfig_standard;
		UA_ServerNetworkLayer nl =
		UA_ServerNetworkLayerTCP(UA_ConnectionConfig_standard, 16664);
		config.networkLayers = &nl;
		config.networkLayersSize = 1;
		UA_Server *server = UA_Server_new(config);
		
		for(int32_t i=1000; i < 10000; i=i+1000) {
				string name = "the.answer_" + to_string(i);
				UA_NodeId nodeId = UA_NODEID_STRING(1, (char*) name.c_str());
				addVariableArray(server, nodeId, i);
		}
		
		/* Update all array*/
		thread* updater = new std::thread(writeVariable, server);
		UA_Server_run(server, &running);
		UA_Server_delete(server);
		nl.deleteMembers(&nl);
		return 0;
		}`

jpfr, 2017-03-19:
Hello @jmrahm ,

It could be that your additional threads are writing into the array at the same time as it gets read by the main server thread. There are two ways to prevent that.

1. Enable have multithreading-support in the stack with the `UA_ENABLE_MULTITHREADING` build option
http://open62541.org/doc/current/building.html#ua-enable-group
2. Use a custom data source where you protect the read/write accesses with a mutex
http://open62541.org/doc/current/tutorial_server_datasource.html#variable-data-sources

When multi-threading issues are positively ruled out, we can have a closer look at the outliers.

jpfr, 2017-03-19:
And thank you for the very complete bug report with code examples and all!

jmrahm, 2017-03-23:
Hello @jpfr ,

thanks for the very fast reply.
I tried some more test. The multithreading option don't solve it. Nevertheless the current 0.2 branch does, beforehand I used the e0.2-rc2 Tag. :-) 

Hence it is solved.

closed 2017-03-23 by jmrahm