open62541/open62541 #998: Something weird is happening with the address space after 49f763441c119313d52874424e7376d497b09c37

glima, 2017-03-24:
Description
===========

Hey, guys.

I've spent some hours with the issue already, but since the code base is a bit too big I wanted to ask for your help in parallel as well with it. Basically, before the aforementioned commit, the code below (I tried to attach it but GitHub won't accept .c files) ran fine—property 'on' of those objects had type LightTypePropState (only one occurrence of the HasTypeDefinition relation appeared on UaExpert). After that commit, those relations appear twice! I'm trying to figure out what is happening. One more thing I also noted is that transitioning between the two commits and having the same UaExpert instance trying to connect to the opc server makes it go crazy: lots of bogus object IDs are shown on the address space. This is not a big deal, though, since having a new UaExpert instance does not lead to that.

Background Information / Reproduction Steps
===========================================

See here for the kinds of information we need to reproduce the bug and how to obtain it:
https://github.com/open62541/open62541/wiki/Writing-Good-Issue-Reports

Compile the sample server and check the exposed address space on UaExpert for both open62541 revisions (I'm also open to suggestions/critique on the lib usage as well—this is a quick demo/test code).

```
#ifdef UA_NO_AMALGAMATION
# include <time.h>
# include "ua_types.h"
# include "ua_server.h"
# include "ua_config_standard.h"
# include "ua_network_tcp.h"
# include "ua_log_stdout.h"
#else
# include "open62541.h"
#endif

#include <errno.h>
#include <linux/limits.h>
#include <signal.h>
#include <stdio.h>
#include <sys/time.h>
#include <time.h>

static bool running = true;

#define LIGHT_TYPE_ID                         50000
#define LIGHT_INSTANCE_ID                     70000
#define NUM_LIGHTS (16)
static UA_NodeId light_obj_id[NUM_LIGHTS];
static UA_NodeId light_type_id;
static UA_NodeId light_state_var_obj_id;
static UA_NodeId light_power_var_obj_id;
static UA_NodeId light_state_var_type_id;
static UA_NodeId light_power_var_type_id;

static size_t light_prop_cnt;
typedef struct {
    UA_Boolean state_val;
    UA_NodeId state_id;
    UA_Double power_val;
    UA_NodeId power_id;
} light_instance;
static light_instance lights[NUM_LIGHTS];

#define LIGHT_TYPE_STATE_VAL_ID               60001
#define LIGHT_TYPE_POWER_VAL_ID               60002

static char *const LIGHT_TYPE_NAME = "LightType";
static char *const LIGHT_TYPE_DESC = "A light";
static char *const LIGHT_TYPE_PROP_STATE_NAME = "LightTypePropState";
static char *const LIGHT_TYPE_PROP_POWER_NAME = "LightTypePropPower";
static char *const LIGHT_TYPE_PROP_STATE_DESC = "Current state";
static char *const LIGHT_TYPE_PROP_POWER_DESC = "Current power level [0.0-1.0]";

// All our temperatures here in Kelvin
// TODO: have more granularity for them (use floating point) and also
// have converters on OCF side to expose in Celsius, later
#define AIR_COND_TYPE_ID                      50001
#define AIR_COND_INSTANCE_ID                  70050
[... 771 lines cut ...]

    ret = UA_Server_run(server, &running);
    UA_Server_removeRepeatedJob(server, jid);

  err_server:
    UA_Server_delete(server);

  err_nl:
    nl.deleteMembers(&nl);

    return ret;
}
```

Checklist
=========
Please provide the following information:

 - [✓ ] open62541 Version (release number or git tag):
 - [NA ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system: Linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [NA ] Wireshark network dump attached
 - [✓ ] Self-contained code example attached
 - [✓ ] Critical issue

jpfr, 2017-03-27:
Hello @glima ,

thanks for the outstandingly complete issue report.
Fixed in 7e4c7e4.

closed 2017-03-27 by jpfr

glima, 2017-03-27:
Thanks a lot for the quick fix as well! There are other funny things happening on UaExpert that might be issues as well, after the same change. Right after the server is brought up, take a look at the value of the light objects' 'on' value—they are Null! Before that change, they reported the 'false' value initially, as they should. Now if I do write those values from a client, they do report boolean values. Any clues?

jpfr, 2017-03-27:
> Right after the server is brought up, take a look at the value of the light objects' 'on' value—they are Null! Before that change, they reported the 'false' value initially, as they should. Now if I do write those values from a client, they do report boolean values. Any clues?

Fixed on master. You need to set a default value for the variabletype. Then, new instances with no initial value get a copy of the variabletype's value.

jpfr, 2017-03-27:
And this is the change for adding the variable types:

```c
static UA_NodeId
define_var_type_inherit(const char *desc,
                        const char *disp_name,
                        const UA_NodeId parent_var_type,
                        UA_UInt32 required_id)
{
    UA_StatusCode ret;
    UA_NodeId ret_id = {0};
    UA_VariableTypeAttributes var_type_attr;
    UA_NodeId type;
    const UA_DataType *typedefinition;

    ret = UA_Server_readDataType(server, parent_var_type, &type);

    typedefinition = UA_findDataType(&type);

    UA_VariableTypeAttributes_init(&var_type_attr);
    var_type_attr.description = UA_LOCALIZEDTEXT("en_US", (char *)desc);
    var_type_attr.displayName = UA_LOCALIZEDTEXT("en_US", (char *)disp_name);
    var_type_attr.dataType = type;
    var_type_attr.valueRank = -1;
    UA_Variant_setScalar(&var_type_attr.value, UA_new(typedefinition), typedefinition);

    UA_Server_addVariableTypeNode
        (server,
         UA_NODEID_NUMERIC(1, required_id),
         UA_NODEID_NUMERIC(0, UA_NS0ID_BASEVARIABLETYPE),
         UA_NODEID_NUMERIC(0, UA_NS0ID_HASSUBTYPE),
         UA_QUALIFIEDNAME(1, (char *)disp_name),
         parent_var_type,
         var_type_attr,
         NULL,
         &ret_id);

    return ret_id;
}
```

glima, 2017-03-27:
> And this is the change for adding the variable types:
>
> ```c
> static UA_NodeId
> define_var_type_inherit(const char *desc,
>                         const char *disp_name,
>                         const [...]

Thanks, Julius!

Can we have my example—or something similar—on the examples dir for
the project? Just to start with, there is no mention of
UA_findDataType() on the ones there now.

Regards,

> 
> https://github.com/open62541/open62541/issues/998#issuecomment-289597472

glima, 2017-03-28:
Another thing that is puzzling me, correct me if I'm wrong:
at `inst_cb()`, `obj_type_id` comes as 60001 (`LIGHT_TYPE_STATE_VAL_ID`), but shouldn't it be 8000 (so that the comparison holds true)? I'm almost sure this once worked.

glima, 2017-03-28:
OK, it seems the type is now `light_state_var_type_id`, works for me.

commit 7e4c7e42 (2017-03-27): fix #998; instantiated child-nodes retained their original references