open62541/open62541 #1004: Duplicate node appear when using xml file to generate .c .h file

xty19, 2017-04-06:
Description
===========
i use UaModeler to generate xml file and use open62541 to generate .c and .h file to add variables
ps: i use UaModeler define my own BaseObjectType  "DeviceObjectType" in  BaseObjectType of ObjectTypes, and i create an instance  "Folder" in Objects folder . But when i add it to server and use UaExpert to browse the varibales , it shows duplicate variables! 
[image]
[image]

Background Information / Reproduction Steps
===========================================

See here for the kinds of information we need to reproduce the bug and how to obtain it:
https://github.com/open62541/open62541/wiki/Writing-Good-Issue-Reports

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag): v0.2.0-RC1
 - [ ] Other OPC UA SDKs used (client or server): server
 - [ ] Operating system:81~precise1-Ubuntu SMP
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2017-04-06:
This looks like an artefact of the node instantiation.
Probably you have the child variables defined for the objecttypes.
The stack automatically instantiates the children together with the object.

And then, the children get added a second time via the nodeset xml model.

@open62541/open62541-core @ichrispa 
Are you okay with adding an additional check for Mandatory and Optional variables for 0.2?
This is defined via an additional references in the objecttype.
If nothing is defined, we then assume "Optional" and don't automatically instantiate children.
Otherwise, this problem will persist throughout the 0.2 release series.

Pro, 2017-04-06:
I ran into the same problem some months ago, where I imported the devices profile via xml into open62541. There I did some hardcoding to fix it, but your proposed is definitely better.

xty19, 2017-04-07:
@jpfr 
its ok to add variables. for 0.2
it goes wrong just adding diy objecttype.

[

](url)

jpfr, 2017-04-14:
Fixed with ee5361074746b2eb125921edb049f6d24cb70ce0.
We now auto-instantiate only those children that are marked as mandatory.

closed 2017-04-14 by jpfr