open62541/open62541 #1006: Help with Method Nodes

syvaren, 2017-04-07:
Description
===========
I have an OPCUA Server setup where I have defined an interface for my service.  I see that UA_Client_addMethodNode can create a Method Node in the Server, but it does not take a callback as a parameter.  How do I register my client's function with the Method Node in the server?

  Client                                                    Server                                                 Client
Function<------------------------------ Register <--------------------------------Call
                                                           MethodNode

Background Information / Reproduction Steps
===========================================

See here for the kinds of information we need to reproduce the bug and how to obtain it:
https://github.com/open62541/open62541/wiki/Writing-Good-Issue-Reports

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag): 0.2 Release
 - [ ] Other OPC UA SDKs used (client or server): 
 - [] Operating system: Debian 7
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Stasik0, 2017-04-07:
How do you plan to transfer the executed code over the network? Sshared object as a blob? Some byte code? Many options here...

syvaren, 2017-04-07:
I had hoped that it would function as more of a forward.  The server would redirect the parameters to the client for processing and then forward the output back.

But these implementation details are more specific to the open62541 framework, than my code right?  How does the framework handle it now?  Again there is a UA_Client_addMethodNode, what is the point of the function if there is no mechanism for calling?

Stasik0, 2017-04-07:
The client probably has macros to add all existing node types...

jpfr, 2017-04-07:
Haha. We asked the same question internally.
The possibility of added MethodeNodes is in the standard. So we support it in the client similar to all the other node types.

Maybe some server has internally a way to run a "constructor" based on the name and arguments of the method. But there is no way to attach executable code via the addnodes service per se.

Stasik0, 2017-04-07:
Afaik no deeper sense here

syvaren, 2017-04-07:
Ok so the ability to add a MethodNode from the client is a red herring then.  So if I wanted to accomplish this I would need to use an Aggregating Server pattern?  Or is there a less complex work around?

syvaren, 2017-04-07:
I was toying with setting the MethodNode as a monitored item and then attempting to handle it in that callback, but I haven't been successful so far.  Is there a way to hijack the input arguments from a MethodNode and write back the outputs?  I realize I am entering dirty hack territory.

Stasik0, 2017-04-07:
Afaik the children of method nodes do not have a value

syvaren, 2017-04-07:
Thanks guys, I appreciate the help.  Keep up the good work.

closed 2017-05-03 by syvaren