open62541/open62541 #1009: UA_Client_NamespaceGetIndex() issue

vdeters, 2017-04-10:
Description
===========
What does it mean when UA_Client_NamespaceGetIndex() returns a bad return code with the following explanation: "The response message size exceeds limits set by the client." I have also seen "An internal error occurred as a result of a programming or configuration error." in response to the same call. I have verified that the namespace I'm passing is valid, because it shows up as the first array entry when I read node UA_NS0ID_SERVER_NAMESPACEARRAY.

Background Information / Reproduction Steps
===========================================

See here for the kinds of information we need to reproduce the bug and how to obtain it:
https://github.com/open62541/open62541/wiki/Writing-Good-Issue-Reports

Checklist
=========
Please provide the following information:

 - [0.2-rc2] open62541 Version (release number or git tag):
 - [server is opcua.demo-this.com:51210/UA/SampleServer] Other OPC UA SDKs used (client or server): 
 - [Ubuntu] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Stasik0, 2017-04-10:
could you capture a wireshark dump? probably a handshake error with the remote server

vdeters, 2017-04-11:
I have a workaround for the time being, which is to get the namespace array
and extract the array index of the matching namespace uri. I could capture
a wireshark dump later, but the issue should be easy to reproduce - just
run a linux client, connect to endpoint "
opcua.demo-this.com:51210/UA/SampleServer", and pass "
http://opcfoundation.org/UA/" to UA_Client_NamespaceGetIndex(). I'm
running 0.2-rc2.
Vince

Stasik0, 2017-04-11:
UA_Client_NamespaceGetIndex() is actually doing exactly the same, can you any differences in the code?

Pro, 2017-04-11:
Can you post the code you use to call the UA_Client_NamespaceGetIndex()? Even better if you post a minimal working example.

It could also be that the way the UA_String is initialized is not correct and therefore there's a serializing error which leads to the `Message size exceeds limits` error.

AGIsmail, 2017-04-13:
I took a moment to mess with this and see if I can reproduce the issue.
I changed the log level to 100 for this and I could not reproduce.

The example c file is here (for now):
https://github.com/AGIsmail/open62541/blob/0.2/examples/client_getNsIndex.c

Client logs: https://paste2.org/AsbFnNvf
Server logs: https://paste2.org/U1G8W1JE

Stasik0, 2017-04-13:
Did you try it with opc.tcp://opcua.demo-this.com:51210/UA/SampleServer ?

AGIsmail, 2017-04-13:
Yes.  I updated the [code](https://github.com/AGIsmail/open62541/blob/0.2/examples/client_getNsIndex.c) to connect to opclabs' public server.

Could not reproduce.

Client logs are here: https://paste2.org/kCaUGkep

vdeters, 2017-04-13:
Added your code to my client and got the segfault below. How do I enable
client logging?

the value is: http://opcfoundation.org/UA/
Testing UA_Client_NamespaceGetIndex() for http://opcfoundation.org/UA/

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff4c3a1f6 in __memcpy_ssse3_back () from /lib64/libc.so.6
(gdb) where
#0  0x00007ffff4c3a1f6 in __memcpy_ssse3_back () from /lib64/libc.so.6
#1  0x00007fffe4b74a2b in Array_encodeBinary ()
#2  0x00007fffe4b74b88 in UA_encodeBinaryInternal ()
#3  0x00007fffe4b74b88 in UA_encodeBinaryInternal ()
#4  0x00007fffe4b74ac6 in Array_encodeBinary ()
#5  0x00007fffe4b74be9 in UA_encodeBinaryInternal ()
#6  0x00007fffe4b79c6f in UA_encodeBinary ()
#7  0x00007fffe4b7b16b in UA_SecureChannel_sendBinaryMessage ()
#8  0x00007fffe4b84f88 in __UA_Client_Service ()
#9  0x00007fffe4b8610d in UA_Client_NamespaceGetIndex ()
#10 0x00007fffe4b71d60 in dfESPopcuaConnector::start (this=0x676190)

AGIsmail, 2017-04-13:
I tested as is with valgrind and got no memory issues at all.
For the sake of doing things a little more right anyway, I changed the array access to use a loop.
The code should now loop over all the URIs returned for the namespaceArray node and get the index for each one.
Tested with valgrind, no leaks, no seg faults.
[Care to try it again?](https://github.com/AGIsmail/open62541/blob/0.2/examples/client_getNsIndex.c)

vdeters, 2017-04-21:
Sorry for the delay...still getting the same crash with your test code:

Reading the value of node (0, UA_NS0ID_SERVER_NAMESPACEARRAY):
Testing UA_Client_NamespaceGetIndex() using the URI http://opcfoundation.org/UA/ 

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff4c371f6 in __memcpy_ssse3_back () from /lib64/libc.so.6
(gdb) where
#0  0x00007ffff4c371f6 in __memcpy_ssse3_back () from /lib64/libc.so.6
#1  0x00007fffe46693fb in Array_encodeBinary ()
#2  0x00007fffe4669558 in UA_encodeBinaryInternal ()
#3  0x00007fffe4669558 in UA_encodeBinaryInternal ()
#4  0x00007fffe4669496 in Array_encodeBinary ()
#5  0x00007fffe46695b9 in UA_encodeBinaryInternal ()
#6  0x00007fffe466e63f in UA_encodeBinary ()
#7  0x00007fffe466fb3b in UA_SecureChannel_sendBinaryMessage ()
#8  0x00007fffe4679958 in __UA_Client_Service ()
#9  0x00007fffe467aadd in UA_Client_NamespaceGetIndex ()
#10 0x00007fffe466684d in dfESPopcuaConnector::start (this=0x67a1a0)

vdeters, 2017-04-21:
One correction, my OS is RHEL 7.1, not Ubuntu.

vdeters, 2017-04-21:
More info...I thought the crash might be related to a failed malloc so I added this after the malloc() call:                cout << "nsUriData = " << (void *)nsUriData << endl;
Now it doesn't crash (huh?)....but the output still shows UA_Client_NamespaceGetIndex() failing as originally reported above:

Reading the value of node (0, UA_NS0ID_SERVER_NAMESPACEARRAY):
nsUriData = 0x687c00
Testing UA_Client_NamespaceGetIndex() using the URI http://opcfoundation.org/UA/ 
Testing UA_Client_NamespaceGetIndex(): Bad status code returned 80020000
Testing UA_Client_NamespaceGetIndex() using the URI urn:opcua.demo-this.com:UA Sample Server 
Testing UA_Client_NamespaceGetIndex(): Bad status code returned 80020000
Testing UA_Client_NamespaceGetIndex() using the URI http://test.org/UA/Data/ 
Testing UA_Client_NamespaceGetIndex(): Bad status code returned 80020000
Testing UA_Client_NamespaceGetIndex() using the URI http://test.org/UA/Data//Instance 
Testing UA_Client_NamespaceGetIndex(): Bad status code returned 80020000
Testing UA_Client_NamespaceGetIndex() using the URI http://opcfoundation.org/UA/Boiler/ 
Testing UA_Client_NamespaceGetIndex(): Bad status code returned 80020000
Testing UA_Client_NamespaceGetIndex() using the URI http://opcfoundation.org/UA/Boiler//Instance 
Testing UA_Client_NamespaceGetIndex(): Bad status code returned 80020000
Testing UA_Client_NamespaceGetIndex() using the URI http://opcfoundation.org/UA/Diagnostics 
Testing UA_Client_NamespaceGetIndex(): Bad status code returned 80020000
Testing UA_Client_NamespaceGetIndex() using the URI http://samples.org/UA/memorybuffer 
Testing UA_Client_NamespaceGetIndex(): Bad status code returned 80020000
Testing UA_Client_NamespaceGetIndex() using the URI http://samples.org/UA/memorybuffer/Instance 
Testing UA_Client_NamespaceGetIndex(): Bad status code returned 80020000

vdeters, 2017-04-24:
Found it...I'm running 0.2-rc2 which does not have this fix:
https://github.com/open62541/open62541/commit/a5c85f03bebb3e5116e947d6dcb97b0d602e6bcf

I copied the current UA_Client_NamespaceGetIndex() code from the 0.2 branch into my client and it works. If I remove UA_ReadValueId_init(&id) (which the above fix added), I get my crash. Closing.

closed 2017-04-24 by vdeters