open62541/open62541 #1010: ExpandedNodeId and reference to another server.

IzonFreak, 2017-04-11:
Description
===========

Hello, 

How do I add another OPC UA server to my own OPC UA server list ? So I can reference it using the Index. 

How do I know the index of my servers ? can this be done at runtime ? 

How can I use ExpandedNodeId to create a reference to a node that is not in my server? 

I was able to use ExpandedNodeId to create a reference to a node in my server using index 0 in the server list. 

Background Information / Reproduction Steps
===========================================

See here for the kinds of information we need to reproduce the bug and how to obtain it:
https://github.com/open62541/open62541/wiki/Writing-Good-Issue-Reports

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2017-04-16:
Hello @IzonFreak,

the variable /Objects/Server/ServerArray holds an array of URIs.

> How do I add another OPC UA server to my own OPC UA server list ? So I can reference it using the Index.

You can use `UA_Server_writeValue` to just overwrite the array.

> How do I know the index of my servers ? can this be done at runtime ?

The server's own uri always has index 0. All other indices can be read at runtime for the variable.

> How can I use ExpandedNodeId to create a reference to a node that is not in my server?

Just set the serverindex to the corresponding value. But the server SDK does currently not act on this information. This is up to the client who reads ist.

jpfr, 2017-04-20:
Closed for now.
Please reopen if questions remain.

closed 2017-04-20 by jpfr