open62541/open62541 #1011: cost too many time when cannt connect server

xty19, 2017-04-17:
Description
===========

when i consider some exception problem, the  time of conduct time has to be counted.

` UA_StatusCode retval = UA_Client_getEndpoints(client, "opc.tcp://localhost:16664",
                                                  &endpointArraySize, &endpointArray);
`

when i run code above it cost nearly 20sec, and then i can check out the server is down.
how to find out the server status raspidly? guys？

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2017-04-17:
Hey @xty19 ,

there is a field "timeout" in the client config.
It is set to 5 seconds by default.

Just use a timeout that suits your needs.

Okay to close?

xty19, 2017-04-18:
@jpfr i set timeout to 500 
but it still cost nearly 20s to return the failure info

xty19, 2017-04-18:
[image]

even i change time out 5000 to 100, it still was 20sec to wait

xty19, 2017-04-20:
@jpfr problem still exists :cry:

jpfr, 2017-04-20:
I tried to find if there is an issue in the client code.
But you need to help us find it.

- Can you enable more logging output by setting UA_LOGLEVEL to 100?
- The client can only "sleep" in the function UA_Connection_receiveChunksBlocking.
  Can you break with a debugger within those 20 seconds?
   a) What is the current call stack. How was the UA_Connection_receiveChunksBlocking reached?
   b) What is the value of the timeout variable?

xty19, 2017-04-20:
@jpfr  ok
tomorrow i will try to find some details :ok_hand:

xty19, 2017-04-24:
@jpfr， i set 100 ， but It still no log

jpfr, 2017-04-24:
In the screenshort above, there is the line `warning/network connection failed`.
That is log output. So logging is definitely working for you.

Can you check what settings you use in the the server config?
Did you recompile the open62541 library after setting UA_LOGLEVEL?

Pro, 2017-08-13:
Fixed with #1154

closed 2017-08-13 by Pro