open62541/open62541 #1012: Calling select fails on Windows after calling shutdown on client socket

bostjanv, 2017-04-18:
Description
===========
When a client closes the connection unexpectedly, shutdown is called on client socket in function ServerNetworkLayerTCP_closeConnection. However, the socket is not removed from the set of sockets to be taken into account in next select call in function ServerNetworkLayerTCP_getJobs. This apparently causes some problems (at least on) Windows platform. (Currently I didn't bother to check for the same issue on Linux.) Specifically, select returns SOCKET_ERROR (-1) with error code set to WSAENOTSOCK 
https://msdn.microsoft.com/en-us/library/windows/desktop/ms740668(v=vs.85).aspx#WSAENOTSOCK

Possible solution
===========
A possible solution could be to remove the socket from the "working" set before the next call to select.

jpfr, 2017-04-19:
Hrmpf. The Windows POSIX API is always "close but not entirely compatible".

Does this PR fix your issue?
https://github.com/open62541/open62541/pull/1013

bostjanv, 2017-04-20:
I didn't have time to test the PR but I'm quite confident that it fix my issue.

Thanks for the fix,
Bostjan

P.S. Please check also issue [https://github.com/open62541/open62541/issues/1015](#1015)

closed 2017-04-20 by jpfr