open62541/open62541 #1015: closeSession callback is not being called

bostjanv, 2017-04-20:
Description
===========
I'm not sure what is the current expected behavior but I have identified two cases where closeSession callback is not being called (but it should have been in my opinion):
- When the session is removed due to timeout.
- When the client closes the connection unexpectedly.

And a somehow related question: Is it possible to renew/reconnect a closed session?

Best regards,
Bostjan

jpfr, 2017-04-20:
Thanks for looking into this.
In general, we have 
```
connection < securechannel < session
```

When the connection / securechannel break, we can recover a detached session from a new securechannel.
But it is not possible to recover a session once it is closed.

> but I have identified two cases where closeSession callback is not being called

You mean the session management that was recently added to master?

> When the session is removed due to timeout

Yes. That needs to be added.

> When the client closes the connection unexpectedly

We will wait until the session times out. So a client can recover the session from a new securechannel.

bostjanv, 2017-04-20:
> When the connection / securechannel break, we can recover a detached session from a new securechannel. But it is not possible to recover a session once it is closed.
Could you please explain this in a little more detail? Maybe you could give me a possible scenario when this could happen. Do we have any example code how the client can recover a detached session?

> You mean the session management that was recently added to master?
Yes.

> Yes. That needs to be added.
Ok, if I understand this correctly, this feature is scheduled for the 0.3 branch? Just to be sure that we are on the same page - current behavior is that the closeSession callback *is* being called when the client closes the connection/session according to the rules but *is not* being called when the session is closed due to timeout.

jpfr, 2017-04-20:
> Could you please explain this in a little more detail? Maybe you could give me a possible scenario when this could happen. 

Say you have a client that wants a complete history of a variable according to some MonitoredItem / sample frequency. When your connection drops out for some time (or you PC crashes), the SecureChannel is lost.
But you can get back the Session via the `ActivateSession` service. Details of that service are in Part 4 of the Spec.

> Ok, if I understand this correctly, this feature is scheduled for the 0.3 branch?

Jup. The callback is a new addition to the master branch. And the entire session management plugin API must be solid before 0.3.

> Just to be sure that we are on the same page - current behavior is that the closeSession callback is being called when the client closes the connection/session according to the rules but is not being called when the session is closed due to timeout.

Yes. You are free to post a PR that adds the callback to the appropriate place.
If you already know your way around in the code, that should be doable in little time.

Otherwise, one of the @open62541/open62541-core members will get to it. I marked the issue for 0.3. So it will be resolved before an eventual release.

jpfr, 2017-05-20:
Since nobody stepped forward, fixed in 691994f.

commit 691994fc (2017-05-20): fix #1015, call accesscontrol.closeSession when a session times out
closed 2017-05-20 by jpfr