open62541/open62541 #1016: Client session closed

vdeters, 2017-04-20:
Description
===========
UA_Client_writeValueAttribute() fails with "The session was closed by the client."

Background Information / Reproduction Steps
===========================================
I have a client that connects, reads a value attribute, and then writes that value attribute once per second indefinitely. After 450 seconds (7.5 minutes) the write fails as described above. It's always 450 seconds after connecting (i.e. 450 writes) - is there something magical about that value? Also, the writes are done on a separate thread from the connect and initial read - does that matter?

On a side note, are log messages supposed to show up on the console by default? I saw nothing so added UA_Logger logger = UA_Log_Stdout; and changed the log level from 300 to 100, but no change.

Checklist
=========
Please provide the following information:

 - [ 0.2-rc2 ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ Ubuntu ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

vdeters, 2017-04-20:
Update: this doesn't happen when connected to a different server, so I guess it's the server disconnecting the client. I'm assuming this is just server policy, so not worried about it.

However the "...closed by the client" part of the error message threw me off - shouldn't it say "...closed by the server"?

Also, would still appreciate help in getting logger output to the console, thanks.

jpfr, 2017-04-20:
The server should give you a reason why it closed the connection.
Did we fail to repeat the SecureChannel handshake? Usually this is done automatically whenever you make a service call and we are over 75% of the handshake interval.

We would be grateful if you could tell us either
- the server-side logged reason for closing the connection
- the server vendor and version. So we can reproduce the condition.
(Feel free to send this over a private channel if you feel more comfortable not to publicly share this.)

> However the "...closed by the client" part of the error message threw me off - shouldn't it say "...closed by the server"?

True. That's misleading. The message comes from the definition of the StatusCode:
```{UA_STATUSCODE_BADSESSIONCLOSED, "BadSessionClosed", "The session was closed by the client."}```
Maybe we should limit the use of BadSessionClosed to the server.

> Also, would still appreciate help in getting logger output to the console, thanks.

Have a look at /plugins/ua_log_stdout.c. On Ubuntu this should work out-of-the-box.
And how did you get the "...closed by the client" part without logging?

vdeters, 2017-04-21:
I'm using a remote public server (opc.tcp://opcua.demo-this.com:51210/UA/SampleServer), so not sure how to access its logs.

About my client logging, I started adding code from ua_log_stdout.c to my client but then discovered that symbol UA_Log_Stdout is already defined somewhere in the library. I just added UA_Logger logger = UA_Log_Stdout; to my code and set this logger in my  UA_ClientConfig, but still not seeing anything sent to stdout. I do see output from other printf statements in my code.

I got the error message by calling UA_StatusCode_description(retval)->explanation after getting the bad retval returned by UA_Client_writeValueAttribute().

vdeters, 2017-04-27:
Closing, appears to be fixed by https://github.com/open62541/open62541/commit/f06f2e8b2023febd9e62ecddd36043ba415d34c2.

closed 2017-04-27 by vdeters