open62541/open62541 #1017: Crash in UA_Client_Subscriptions_manuallySendPublishRequest()

vdeters, 2017-04-21:
Description
===========
Calling UA_Client_Subscriptions_manuallySendPublishRequest() in a tight loop, since it seems to block anyway. After about 15 minutes (without the handler being called, because there are no new values), I get this crash:

*** Error in `/mnt/data/home/videte/esp-examples/cxx/subscribe_server/./l_build/lin64/subscribe_server': double free or corruption (fasttop): 0x00000000006822c0 ***
Program received signal SIGABRT, Aborted.
[Switching to Thread 0x7fffcffff700 (LWP 128796)]
0x00007ffff4b225d7 in raise () from /lib64/libc.so.6
Missing separate debuginfos, use: debuginfo-install glibc-2.17-78.el7.x86_64 keyutils-libs-1.5.8-3.el7.x86_64 krb5-libs-1.12.2-14.el7.x86_64 libcom_err-1.42.9-7.el7.x86_64 libselinux-2.2.2-6.el7.x86_64 libuuid-2.23.2-22.el7_1.1.x86_64 nss-softokn-freebl-3.16.2.3-12.el7_1.x86_64 numactl-libs-2.0.9-5.el7_1.x86_64 openssl-devel-1.0.1e-42.el7_1.9.x86_64 openssl-libs-1.0.1e-42.el7_1.9.x86_64 pcre-8.32-14.el7.x86_64 xz-libs-5.1.2-9alpha.el7.x86_64 zlib-1.2.7-13.el7.x86_64
(gdb) where
#0  0x00007ffff4b225d7 in raise () from /lib64/libc.so.6
#1  0x00007ffff4b23cc8 in abort () from /lib64/libc.so.6
#2  0x00007ffff4b62e07 in __libc_message () from /lib64/libc.so.6
#3  0x00007ffff4b6a1fd in _int_free () from /lib64/libc.so.6
#4  0x00007fffe466b273 in deleteMembers_noInit ()
#5  0x00007fffe466b273 in deleteMembers_noInit ()
#6  0x00007fffe466b311 in UA_deleteMembers ()
#7  0x00007fffe46793da in SecureChannelHandshake ()
#8  0x00007fffe467972e in UA_Client_manuallyRenewSecureChannel ()
#9  0x00007fffe4679774 in __UA_Client_Service ()
#10 0x00007fffe467c97b in UA_Client_Subscriptions_manuallySendPublishRequest()
#11 0x00007fffe465f3dc in dfESPopcuaConnector::pubThread (this=0x7fffcffee448)

Background Information / Reproduction Steps
===========================================

See here for the kinds of information we need to reproduce the bug and how to obtain it:
https://github.com/open62541/open62541/wiki/Writing-Good-Issue-Reports

Checklist
=========
Please provide the following information:

 - [ 0.2-rc2 ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ RHEL 7.1 ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

vdeters, 2017-04-21:
Another data point: I get a very similar looking crash when I call UA_Client_delete () after getting an error code back from UA_Client_writeValueAttribute():

*** Error in `/mnt/data/home/videte/esp-examples/cxx/subscribe_server/./l_build/lin64/subscribe_server': double free or corruption (fasttop): 0x0000000000686c70 ***
Program received signal SIGABRT, Aborted.
[Switching to Thread 0x7fffcffff700 (LWP 45784)]
0x00007ffff4b225d7 in raise () from /lib64/libc.so.6
Missing separate debuginfos, use: debuginfo-install glibc-2.17-78.el7.x86_64 keyutils-libs-1.5.8-3.el7.x86_64 krb5-libs-1.12.2-14.el7.x86_64 libcom_err-1.42.9-7.el7.x86_64 libselinux-2.2.2-6.el7.x86_64 libuuid-2.23.2-22.el7_1.1.x86_64 nss-softokn-freebl-3.16.2.3-12.el7_1.x86_64 numactl-libs-2.0.9-5.el7_1.x86_64 openssl-devel-1.0.1e-42.el7_1.9.x86_64 openssl-libs-1.0.1e-42.el7_1.9.x86_64 pcre-8.32-14.el7.x86_64 xz-libs-5.1.2-9alpha.el7.x86_64 zlib-1.2.7-13.el7.x86_64
(gdb) where
#0  0x00007ffff4b225d7 in raise () from /lib64/libc.so.6
#1  0x00007ffff4b23cc8 in abort () from /lib64/libc.so.6
#2  0x00007ffff4b62e07 in __libc_message () from /lib64/libc.so.6
#3  0x00007ffff4b6a1fd in _int_free () from /lib64/libc.so.6
#4  0x00007fffe466c033 in deleteMembers_noInit ()
#5  0x00007fffe466c0d1 in UA_deleteMembers ()
#6  0x00007fffe467af07 in UA_Client_deleteMembers ()
#7  0x00007fffe467b519 in UA_Client_delete ()
#8  0x00007fffe46615cf in dfESPopcuaConnector::stop (this=0x67de40, 
    this@entry=0x0, inCallback=inCallback@entry=true)

vdeters, 2017-04-25:
Upgraded to a current pull from 0.2 and the crash looks a bit different now (see below). It's trying to free a string in the UA_OpenSecureChannelRequest after doing the handshake, but I'm not sure which string. I have another crash in UA_Client_delete() where it can't free client->authenticationToken, and I see that SecureChannelHandshake() copies its token from the client->authenticationToken, so I wonder if my client->authenticationToken is uninitialized - where does that get set?

*** Error in `/mnt/data/home/videte/esp-examples/cxx/subscribe_server/./l_build/lin64/subscribe_server': double free or corruption (fasttop): 0x00000000006822e0 ***
Program received signal SIGABRT, Aborted.
[Switching to Thread 0x7fffcffff700 (LWP 105509)]
0x00007ffff4b205d7 in raise () from /lib64/libc.so.6
Missing separate debuginfos, use: debuginfo-install glibc-2.17-78.el7.x86_64 keyutils-libs-1.5.8-3.el7.x86_64 krb5-libs-1.12.2-14.el7.x86_64 libcom_err-1.42.9-7.el7.x86_64 libselinux-2.2.2-6.el7.x86_64 libuuid-2.23.2-22.el7_1.1.x86_64 nss-softokn-freebl-3.16.2.3-12.el7_1.x86_64 numactl-libs-2.0.9-5.el7_1.x86_64 openssl-devel-1.0.1e-42.el7_1.9.x86_64 openssl-libs-1.0.1e-42.el7_1.9.x86_64 pcre-8.32-14.el7.x86_64 xz-libs-5.1.2-9alpha.el7.x86_64 zlib-1.2.7-13.el7.x86_64
(gdb) where
#0  0x00007ffff4b205d7 in raise () from /lib64/libc.so.6
#1  0x00007ffff4b21cc8 in abort () from /lib64/libc.so.6
#2  0x00007ffff4b60e07 in __libc_message () from /lib64/libc.so.6
#3  0x00007ffff4b681fd in _int_free () from /lib64/libc.so.6
#4  0x00007fffe443970e in String_deleteMembers ()
#5  0x00007fffe4439c9d in NodeId_deleteMembers ()
#6  0x00007fffe443b917 in deleteMembers_noInit ()
#7  0x00007fffe443b917 in deleteMembers_noInit ()
#8  0x00007fffe443b99e in UA_deleteMembers ()
#9  0x00007fffe4437d83 in UA_OpenSecureChannelRequest_deleteMembers ()
#10 0x00007fffe4460788 in SecureChannelHandshake ()
#11 0x00007fffe4461a9a in UA_Client_manuallyRenewSecureChannel ()
#12 0x00007fffe4461e3c in __UA_Client_Service ()
#13 0x00007fffe4438e19 in UA_Client_Service_publis
h ()
#14 0x00007fffe4464416 in UA_Client_Subscriptions_manuallySendPublishRequest ()
#15 0x00007fffe442e3ac in dfESPopcuaConnector::pubThread (this=0x7fffcffee448)

vdeters, 2017-04-25:
> Hello @vdeters ,
>
> commit f06f2e8 should have fixed your issue insofar that UA_OpenSecureChannelRequest_deleteMembers is no longer required within SecureChannelHandshake.
>
> Can you try with the current 0.2 branch?
> It seems you have built quite a test-suite to bring open62541 under load... 😄

Thanks for the fix, I'll try it tomorrow. All I did was set up a subscription and call manuallySendPublishRequest() repeatedly, so I'm wondering why this doesn't affect all clients that use subscriptions.

vdeters, 2017-04-26:
This issue is resolved. Is there any chance of another release in the next few weeks?

jpfr, 2017-04-26:
Definitely.
Once the issues that came up during the last 10 days are solved, we should be good to go.
The policy for releases is "no known bugs and no known warnings/issues in all tests".

We can start by closing this issue. 😄

closed 2017-04-26 by jpfr