open62541/open62541 #1022: UA_NodeStore_newNode does not add node to NodeStore

Pro, 2017-04-27:
Could it be that the `UA_Client_Service_addNodes` service call does not add the newly created node to the nodestore? (Current 0.2 branch)

It calls:
- `Service_AddNodes_single` 
- `createNodeFromAttributes` (https://github.com/open62541/open62541/blob/0.2/src/server/ua_services_nodemanagement.c#L787)
- `UA_NodeStore_newNode` (https://github.com/open62541/open62541/blob/0.2/src/server/ua_services_nodemanagement.c#L732)

As far as I understood, the `UA_NodeStore_newNode` does not add the node to the nodestore?

I created some unit tests (see #1021).
There I first call `UA_Client_addViewNode` https://github.com/open62541/open62541/pull/1021/commits/f11d1cbada3d1080da1e8fa2345425cfe961d779#diff-17293ede5b62a1470244cd496f747cb2R240 and then `UA_Client_Service_deleteNodes` https://github.com/open62541/open62541/pull/1021/commits/f11d1cbada3d1080da1e8fa2345425cfe961d779#diff-17293ede5b62a1470244cd496f747cb2R393 to delete the just created node.

The delete call fails with the statuscode `UA_STATUSCODE_BADNODEIDUNKNOWN` which means that the node is not in the nodestore. If I pass e.g. the Server node to the delete call, it succeeds.

jpfr, 2017-04-27:
> Could it be that the UA_Client_Service_addNodes service call does not add the newly created node to the nodestore?

Once you are in `Service_AddNodes_single`, you take the exact same codepath used to bootstrap ns0.

> There I first call UA_Client_addViewNode f11d1cb#diff-17293ede5b62a1470244cd496f747cb2R240 and then UA_Client_Service_deleteNodes f11d1cb#diff-17293ede5b62a1470244cd496f747cb2R393 to delete the just created node.

What is the return status of addViewNode?

Pro, 2017-04-27:
Return status of addViewNode is Success, so no problem there...

I'll try to create a minimal example to further investigate

Pro, 2017-04-27:
Ok, the problem is related to adding and deleting references.

The following example code outputs `BadNodeIdUnknown` when deleting the view node. If you remove the `UA_Client_deleteReference` call, then the view node is successfully deleted.
So it seems that deleting the reference also deletes the view node, which should not happen.

```
#include <stdio.h>
#include "open62541.h"

int main(void) {
    UA_Client *client = UA_Client_new(UA_ClientConfig_standard);
    UA_StatusCode retval = UA_Client_connect(client, "opc.tcp://localhost:4840");
    if(retval != UA_STATUSCODE_GOOD) {
        UA_Client_delete(client);
        return (int)retval;
    }

    UA_ViewAttributes attr;
    UA_ViewAttributes_init(&attr);
    attr.description = UA_LOCALIZEDTEXT("en_US", "View to Server");
    attr.displayName = UA_LOCALIZEDTEXT("en_US", "Servers");

    UA_NodeId newViewId;

    retval = UA_Client_addViewNode(client, UA_NODEID_NULL,
                                   UA_NODEID_NUMERIC(0, UA_NS0ID_VIEWSFOLDER),
                                   UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES),
                                   UA_QUALIFIEDNAME(1, "Servers"),
                                   attr, &newViewId);
    if(retval != UA_STATUSCODE_GOOD) {
        UA_Client_delete(client);
        printf("ERROR adding view node: %s", UA_StatusCode_name(retval));
        return (int)retval;
    }

    retval = UA_Client_addReference(client, newViewId, UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES),
                                    UA_TRUE, UA_STRING_NULL,
                                    UA_EXPANDEDNODEID_NUMERIC(0, UA_NS0ID_SERVER),
                                    UA_NODECLASS_VARIABLE);

    if(retval != UA_STATUSCODE_GOOD) {
        UA_Client_delete(client);
        printf("ERROR adding reference node: %s", UA_StatusCode_name(retval));
        return (int)retval;
    }

    retval = UA_Client_deleteReference(client, newViewId, UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES),
                                       UA_TRUE, UA_EXPANDEDNODEID_NUMERIC(0, UA_NS0ID_SERVER),
                                       UA_FALSE);

    if(retval != UA_STATUSCODE_GOOD) {
        UA_Client_delete(client);
        printf("ERROR deleting reference node: %s", UA_StatusCode_name(retval));
        return (int)retval;
    }

    retval = UA_Client_deleteNode(client, newViewId, UA_TRUE);

    if(retval != UA_STATUSCODE_GOOD) {
        UA_Client_delete(client);
        printf("ERROR deleting view node: %s", UA_StatusCode_name(retval));
        return (int)retval;
    }

    UA_Client_delete(client); /* Disconnects the client internally */
    return UA_STATUSCODE_GOOD;
}
```

Pro, 2017-04-27:
Funny copy paste error. Fixed https://github.com/open62541/open62541/pull/1021/commits/aa6ec63e4c39885ffd328c0431db00ab648ac175

closed 2017-04-27 by Pro

jpfr, 2017-04-27:
That makes you the bug hunter of the month.
At least! 😏

commit aa6ec63e (2017-04-27): DeleteReferences deleted nodes instead of the reference. Fix #1022
commit 98bf4cad (2017-04-27): DeleteReferences deleted nodes instead of the reference. Fix #1022