open62541/open62541 #1023: is readArrayDimension checking the wrong value?

Pro, 2017-04-28:
Im currently writing some additional unit tests for client highlevel (see https://github.com/open62541/open62541/blob/feature/unit_tests/tests/check_client_highlevel.c#L412).

Maybe I did not yet get the array dimensions thingy right, but I think there may be a problem.

I have an issue with the `writeArrayDimensions` and `readArrayDimensions`. The methods are evaluating the `arrayDimensions` value from the `UA_DataValue.arrayDimensions` field instead of the `UA_VariableNode.arrayDimensions`.

`UA_Client_readArrayDimensionsAttribute` calls:
- `Service_Read_single` https://github.com/open62541/open62541/blob/0.2/src/server/ua_services_attribute.c#L754
- `readArrayDimensionsAttribute` https://github.com/open62541/open62541/blob/0.2/src/server/ua_services_attribute.c#L207
So it returns the `UA_DataValue.arrayDimensions` field but should return `UA_VariableNode.arrayDimensions`

Same for `UA_Client_writeArrayDimensionsAttribute`which calls:
- `Service_Write` -> `UA_Server_editNode` -> `CopyAttributeIntoNode`https://github.com/open62541/open62541/blob/0.2/src/server/ua_services_attribute.c#L1146
- `writeArrayDimensionsAttribute` https://github.com/open62541/open62541/blob/0.2/src/server/ua_services_attribute.c#L1083

The problem is at line https://github.com/open62541/open62541/blob/0.2/src/server/ua_services_attribute.c#L256 where it checks for `value.value.arrayDimensions`
```
        retval = compatibleArrayDimensions(arrayDimensionsSize, arrayDimensions,
                                           value.value.arrayDimensionsSize, value.value.arrayDimensions);
```
I think that here it should use `node.arrayDimensions` instead.

Here is a simple test program demonstrating the problem. Use the `feature/unit_tests` branch for testing since it includes some additional fixes (Based on `0.2`):

```
#include <stdio.h>
#include "open62541.h"

int main(void) {
    UA_Client *client = UA_Client_new(UA_ClientConfig_standard);
    UA_StatusCode retval = UA_Client_connect(client, "opc.tcp://localhost:4840");
    if(retval != UA_STATUSCODE_GOOD) {
        UA_Client_delete(client);
        return (int)retval;
    }

	UA_NodeId nodeArrayId;

	UA_VariableAttributes attr;
	UA_VariableAttributes_init(&attr);
	attr.description = UA_LOCALIZEDTEXT("en_US", "Array");
	attr.displayName = UA_LOCALIZEDTEXT("en_US", "Array");

	UA_Int32 values[2];
	values[1] = 10;
	values[2] = 20;

	UA_Variant_setArray(&attr.value, values, 2, &UA_TYPES[UA_TYPES_INT32]);
	attr.dataType = UA_TYPES[UA_TYPES_INT32].typeId;
	attr.valueRank = 1; /* array with one dimension */
	UA_UInt32 arrayDims[1] = {2};
	attr.arrayDimensions = arrayDims;
	attr.arrayDimensionsSize = 1;

	retval = UA_Client_addVariableNode(client, UA_NODEID_NULL,
									   UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
									   UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES),
									   UA_QUALIFIEDNAME(1, "Array"),
									   UA_NODEID_NULL, attr, &nodeArrayId);
	if(retval != UA_STATUSCODE_GOOD) {
		printf("addVariableNode error: %s", UA_StatusCode_name(retval));
		UA_Client_delete(client);
		return (int)retval;
	}

	UA_UInt32 *arrayDimsRead;
	size_t arrayDimsReadSize;
	retval = UA_Client_readArrayDimensionsAttribute(client, nodeArrayId, &arrayDimsRead , &arrayDimsReadSize);
	if(retval != UA_STATUSCODE_GOOD) {
		printf("UA_Client_readArrayDimensionsAttribute error: %s", UA_StatusCode_name(retval));
		UA_Client_delete(client);
		return (int)retval;
	}
	if (arrayDimsReadSize != 1) {
		printf("arrayDimsReadSize != 1. Val: %ld", arrayDimsReadSize);
		UA_Client_delete(client);
		return 1;
	}
	if (arrayDimsRead[0] != 2) {
		printf("arrayDimsRead[0] != 2. Val: %d", arrayDimsRead[0] );
		UA_Client_delete(client);
		return 1;
	}

	UA_UInt32 arrayDimsNew[] = {3};
	retval = UA_Client_writeArrayDimensionsAttribute(client, nodeArrayId, arrayDimsNew , 1);
	if(retval != UA_STATUSCODE_GOOD) {
		printf("UA_Client_writeArrayDimensionsAttribute error: %s", UA_StatusCode_name(retval));
		UA_Client_delete(client);
		return (int)retval;
	}

	retval = UA_Client_readArrayDimensionsAttribute(client, nodeArrayId, &arrayDimsRead , &arrayDimsReadSize);
	if(retval != UA_STATUSCODE_GOOD) {
		printf("UA_Client_readArrayDimensionsAttribute error: %s", UA_StatusCode_name(retval));
		UA_Client_delete(client);
		return (int)retval;
	}
	if (arrayDimsReadSize != 1) {
		printf("arrayDimsReadSize != 1. Val: %ld", arrayDimsReadSize);
		UA_Client_delete(client);
		return 1;
	}
	if (arrayDimsRead[0] != 3) {
		printf("arrayDimsRead[0] != 3. Val: %d", arrayDimsRead[0] );
		UA_Client_delete(client);
		return 1;
	}

	UA_Client_disconnect(client);
	UA_Client_delete(client);
	printf("I'm done here!");
}
```

The first read already returns an invalid value for the arrayDimension. It returns `0` but should be IMHO `2`?
```
arrayDimsRead[0] != 2. Val: 0
```

jpfr, 2017-04-28:
> readArrayDimensionsAttribute https://github.com/open62541/open62541/blob/0.2/src/server/ua_services_attribute.c#L207
So it returns the UA_DataValue.arrayDimensions field but should return UA_VariableNode.arrayDimensions

The code is
```c
UA_Variant_setArray(&v->value, vn->arrayDimensions,
       vn->arrayDimensionsSize, &UA_TYPES[UA_TYPES_INT32]);
```
So the array dimensions from the variablenode (vn) are returned.

> The problem is at line https://github.com/open62541/open62541/blob/0.2/src/server/ua_services_attribute.c#L256 where it checks for value.value.arrayDimensions

Here, it checks whether the current value-variant is compatible with the array dimensions of the node. Looks legit.

-----

There are three places where array dimensions live.
- In the value variant
- In the variable node
- In the variabletype node

The array dimensions in the variable node are "constraints" for the value variant. And the array dimensions in the variabletype are "constraints" for the variable node.
The constraints may contain entries with zero. That's a wildcard. So every length is permitted for that dimension.

Pro, 2017-04-28:
Thanks for the clarification.
The problem with the readArrayDimensionsAttribute was a read after free. Fixed in:
https://github.com/open62541/open62541/commit/1ace5a706a769639e1a79a595b28c928964ddbdd

The writeArrayDimensionsAttribute still is a problem.

> >    The problem is at line https://github.com/open62541/open62541/blob/0.2/src/serverua_services_attribute.c#L256 where it checks for value.value.arrayDimensions
> Here, it checks whether the current value-variant is compatible with the new array dimensions of the variable node before writing them.

This means that whenever a variable has a value, I can not change the arrayDimensions anymore?
Because the new array dimensions have to match the current dimensions of the value varian. On the other hand dimensions of the value variant have to match the array dimensions of the variable node.

jpfr, 2017-04-28:
> This means that whenever a variable has a value, I can not change the arrayDimensions anymore?

The question is whether it is okay that the constraints are not upheld at some point.
If not, you have to first remove the value alltogether, then change the arraydimensions, then set a new value.

closed 2017-04-28 by Pro