open62541/open62541 #1025: Can not Creating object instances

sunwenfeng, 2017-05-03:
Description
===========

I have successfully created the FIELDDEVICE object type, now I try to run the Creating object instances follow the tutorial. I changed my code with reference to the answer of Yang xinyu. After running(gcc -g -std=c99 -Wl,-rpath,'pwd' -I ./include -L . -DUA_ENABLE_METHODCALLS -o pump_instance ./pump_instance.c -lopen62541), the instance pump1 did not create , can you tell me how you running the code ? What should I do before running?

THANKS！

Background Information / Reproduction Steps
===========================================

See here for the kinds of information we need to reproduce the bug and how to obtain it:
https://github.com/open62541/open62541/wiki/Writing-Good-Issue-Reports

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Pro, 2017-05-03:
Hi @sunwenfeng, thanks for your question.

Can you try to use the most current version from git:
https://github.com/open62541/open62541/tree/0.2

Can you also post example code how you are creating the object type and object instances. Even better if you could post a minimum working example code (https://stackoverflow.com/help/mcve)

jpfr, 2017-05-11:
Hello @sunwenfeng ,

did you add a reference to make the content of the object mandatory?
See also the tutorial here: https://open62541.org/doc/current/tutorial_server_object.html#object-types-type-hierarchies-and-instantiation

YangXinyu, 2017-05-13:
Hello @sunwenfeng ,
i hope you can try the current version or you can try the released version : 
[https://github.com/open62541/open62541/tree/v0.2.0-RC1](url)

it works, i guess.

closed 2017-09-14 by Pro