open62541/open62541 #1027: ByteString length

Infinity95, 2017-05-04:
In the current implementation ByteStrings have their length defined as a size_t.
The Standard specifies, that the length of a ByteString is a 32-bit signed integer:

> 5.2.2.7 ByteString
> A ByteString is encoded as sequence of bytes preceded by its length in bytes. The length is
> encoded as a 32-bit signed integer as described above.
> If the length of the byte string is −1 then the byte string is ‘null’.

What's the reasoning behind the size_t? Is it intentional?

jpfr, 2017-05-04:
> What's the reasoning behind the size_t? Is it intentional?

Yes.
All C libraries worth their salt use size_t for lengths.
See also here: https://ewontfix.com/9/

In OPC UA, there is a difference between an empty array (size 0) and a null array (size -1).
But using signed ints for array sizes is **very risky**. You need tons of if/else checks.
Negative indices, overflowing, and so on.
Furthermore, signed integers should generally not be used in for-loops.
Because signed integers are **undefined** after an overflow. And the compiler cannot apply certain optimizations. https://news.ycombinator.com/item?id=11146384

In open62541, empty arrays and null arrays have both length zero.
The null array uses a special sentinel data-pointer to differentiate between both cases.
The sentinel pointer works for us, because it is never dereferenced anyway.
The behaviour is the same for all arrays. Including strings and bytestrings.
http://open62541.org/doc/current/types.html#array-handling

The change in representation compared to the wire protocol is hidden in the encoding layer.
There, we also check for length <= INT32_MAX, and so on.

closed 2017-05-04 by jpfr

Infinity95, 2017-05-04:
Ok, thanks for clarifying.

jpfr, 2017-05-04:
Here's some more motivation for using size_t.

https://www.securecoding.cert.org/confluence/display/c/INT01-C.+Use+rsize_t+or+size_t+for+all+integer+values+representing+the+size+of+an+object
https://gustedt.wordpress.com/2013/07/15/a-praise-of-size_t-and-other-unsigned-types/