open62541/open62541 #1028: TCP Error message in processChunks

Infinity95, 2017-05-04:
Currently there is functionality for handling TCP error messages in the processChunks function:
https://github.com/open62541/open62541/blob/master/src/ua_securechannel.c#L365

Shouldn't this be moved to a different location?
Or did i get something wrong here?

There also seems to be a function for processing these error messages in ua_server_binary.c:
https://github.com/open62541/open62541/blob/master/src/server/ua_server_binary.c#L535

What i also don't get here is why the Message types for Chunks are mixed with those for TCP messages.
Isn't the type of the tcp messages encoded in the TCP Message header (1.03 Part 6 Table 34) and the type of the chunk in the secure conversation message header (1.03 Part 6 Table 26)?

jpfr, 2017-05-04:
> Shouldn't this be moved to a different location?
> There also seems to be a function for processing these error messages in ua_server_binary.c:

Yes, please. If that's redundant functionality, you are welcome to make a generic implementation available in both places. Or, even better, make error reporting happen only in one of those places.

> What i also don't get here is why the Message types for Chunks are mixed with those for TCP messages.
The message type (e.g. OPN) needs three bytes.
And the chunk type one more byte.

Instead of doing many comparisons
```c
if(buf[0] = 'O' && buf[1] == 'P' && buf[2] == 'N')
```
we just decode the (OPNF) as an unsigned 32bit integer (little-endian), mask out the bits we don't need and do just one integer comparison.

Infinity95, 2017-05-04:
> Instead of doing many comparisons
> 
> `if(buf[0] = 'O' && buf[1] == 'P' && buf[2] == 'N')`
> 
> we just decode the (OPNF) as an unsigned 32bit integer (little-endian), mask out the bits we don't need and do just one integer comparison.
Of course. My point was why store the TCP message type and the SecureConversation message type in the same variable and store the types in the same enum? As far as i understood these have nothing to do with eachother, right? What if in the future there was an overlap in the message types. Wouldn't that be problematic? (Assuming this could be the case one day, but i doubt it.)

jpfr, 2017-05-04:
Do you have an example for such an "overlap"?
If the protocol changes (which it won't in the foreseeable future), we need to touch the code anyway.
I don't see a case we can't cover with the current appraoch.

You can, of course, propose a change.
But let's make this separate from the security policies PR.

It's good to have somebody question current practices. 😄 
Just reopen if necessary.

closed 2017-05-04 by jpfr

Infinity95, 2017-05-04:
It would rather be a matter of cleanliness than fixing a (potential) problem.
I find it cleaner to handle these things separately since they have nothing to do with each other.
Though, this is probably something with low priority which can be approached at a later time.
It's just something i stumbled upon while modifying the code to incorporate the security policies and i found it a bit odd.
I'll investigate if this is reasonable and how much work it would approximately be to change.

jpfr, 2017-05-04:
Ok, cool.