open62541/open62541 #1030: Crashed at end of lifetime for subscription

JinGuoGe, 2017-05-06:
Hi
open62541 version tag - 11b729df23 (https://open62541.org/releases) 
I am using https://github.com/comtel2000/opc-ua-client for OpcUaClient.
I am using twp pcs for test. - One is for comtel2000 OPCUaClient, and then the other is for open62541 based OPCUaServer.
Open62541 is running in Ubuntu environment.

In OPCUaClient side, i sent monitor request

[image]

Then, i plugged off network of open62541 running pc.
After some times (about 5min), then open62541 is crashed with following gdb track.

https://paste2.org/be9PYLWD

When i used eclipse-milo sdk for opcUaClient, it happened same. 
Looking forward your help.
Thanks.

jpfr, 2017-05-08:
Hello @JinGuoGe ,

I am unable to run comtel2000/opc-ua-client.

Neither of
- mvn clean install
- mvn clean compile assembly:single

produces a runable jar (Windows, Java 8, current Maven).
What is your setup to reproduce the issue?

JinGuoGe, 2017-05-08:
Hi,
I just used Eclipse Mars for running comtel2000/opc-ua-client.
Just 3 steps.
1) Importing
[image]
2)Selecting pom file
[image]
3)running
[image]

jpfr, 2017-05-08:
Hello @JinGuoGe ,

can you test with the current 0.2 branch?
Should be fixed by the latest commit.

JinGuoGe, 2017-05-08:
Hi @jpfr 
Thanks for your revision.
It works well.
Very thank you.

closed 2017-05-08 by JinGuoGe