open62541/open62541 #1031: Instantiations of complex objects are missing members.

syvaren, 2017-05-08:
Description
===========

I have defined a complex type as follows:
[image]

It is my understanding that after I have defined this as a structure in OPC UA all that is needed is to instantiate the object and all of its members are created for me.

Here you can see that the definition is viewable in the server.
[image]

But an instance of this object only leaves we with the variable level, it doesn't have the variable's properties.
[image]

I can include some code if requested, it just is a rather large sample, hopefully pictures can adequately explain this issue.  Is this a known bug or am I failing to do something properly?

Background Information / Reproduction Steps
===========================================

See here for the kinds of information we need to reproduce the bug and how to obtain it:
https://github.com/open62541/open62541/wiki/Writing-Good-Issue-Reports

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag): master (HASH: 4f23e1d)
 - [ ] Other OPC UA SDKs used (client or server): None
 - [ ] Operating system: Debian 7
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2017-05-09:
Were the properties in question marked as mandatory?
https://open62541.org/doc/current/tutorial_server_object.html#object-types-type-hierarchies-and-instantiation

syvaren, 2017-05-09:
Yes, they were, in fact that is the tutorial I used to create my model.
[image]

syvaren, 2017-05-09:
I didn't have a lot of time left over yesterday, but I will try to post some sample code that reproduces this issue today.

syvaren, 2017-05-09:
Ok here is some code (hopefully easily paste-able into a main) that demonstrates what I'm seeing.  I guess it might be important to note that the Property Color is optional in the DoorType variable definition, but in the Definition of CarType both Door and Color are mandatory.

 UA_NodeId carTypeId = UA_NODEID_NUMERIC(1, 1002);
    UA_ObjectTypeAttributes carTypeAttributes;
    UA_ObjectTypeAttributes_init(&carTypeAttributes);
    carTypeAttributes.displayName = UA_LOCALIZEDTEXT("en_US", "CarType");
    UA_Server_addObjectTypeNode(
        _server,
        carTypeId,
        UA_NODEID_NUMERIC(0, UA_NS0ID_BASEOBJECTTYPE),
        UA_NODEID_NUMERIC(0, UA_NS0ID_HASSUBTYPE),
        UA_QUALIFIEDNAME(1, "CarType"),
        carTypeAttributes,
        nullptr,
        nullptr
    );

    UA_Int32 value = 0;
    UA_Variant variant;
    UA_Variant_init(&variant);
    UA_Variant_setScalar(&variant, &value, &UA_TYPES[UA_TYPES_INT32]);

    UA_NodeId doorTypeId = UA_NODEID_NUMERIC(1, 2002);
    UA_VariableTypeAttributes doorTypeAttributes;
    UA_VariableTypeAttributes_init(&doorTypeAttributes);
    doorTypeAttributes.displayName = UA_LOCALIZEDTEXT("en_US", "DoorType");
    doorTypeAttributes.value = variant;
    doorTypeAttributes.dataType = UA_NODEID_NUMERIC(0, UA_NS0ID_INT32);
    doorTypeAttributes.valueRank = -1;
    UA_Server_addVariableTypeNode(
        _server,
        doorTypeId,
        UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE),
        UA_NODEID_NUMERIC(0, UA_NS0ID_HASSUBTYPE),
        UA_QUALIFIEDNAME(1, "DoorType"),
        UA_NODEID_NULL,
        doorTypeAttributes,
        nullptr,
        nullptr
    );

    UA_NodeId colorId = UA_NODEID_NUMERIC(1, 6001);
    UA_VariableAttributes colorAttributes;
    UA_VariableAttributes_init(&colorAttributes);
    colorAttributes.displayName = UA_LOCALIZEDTEXT("en_US", "Color");
    UA_Variant_setScalar(&variant, &value, &UA_TYPES[UA_TYPES_INT32]);
    colorAttributes.value = variant;
    colorAttributes.dataType = UA_NODEID_NUMERIC(0, UA_NS0ID_INT32);
    colorAttributes.valueRank = -1;
    UA_Server_addVariableNode(
        _server,
        colorId,
        doorTypeId,
        UA_NODEID_NUMERIC(0, UA_NS0ID_HASPROPERTY),
        UA_QUALIFIEDNAME(1, "Color"),
        UA_NODEID_NUMERIC(0, UA_NS0ID_PROPERTYTYPE),
        colorAttributes,
        nullptr,
        nullptr
    );
    UA_Server_addReference(
        _server,
        colorId,
        UA_NODEID_NUMERIC(0, UA_NS0ID_HASMODELLINGRULE),
        UA_EXPANDEDNODEID_NUMERIC(0, UA_NS0ID_MODELLINGRULE_OPTIONAL),
        true
    );

    UA_NodeId carDoorId = UA_NODEID_NUMERIC(1, 6002);
    UA_VariableAttributes carDoorAttributes;
    UA_VariableAttributes_init(&carDoorAttributes);
    carDoorAttributes.displayName = UA_LOCALIZEDTEXT("en_US", "Door");
    UA_Variant_setScalar(&variant, &value, &UA_TYPES[UA_TYPES_INT32]);
    carDoorAttributes.value = variant;
    carDoorAttributes.dataType = UA_NODEID_NUMERIC(0, UA_NS0ID_INT32);
    carDoorAttributes.valueRank = -1;
    UA_Server_addVariableNode(
        _server,
        carDoorId,
        carTypeId,
        UA_NODEID_NUMERIC(0, UA_NS0ID_HASCOMPONENT),
        UA_QUALIFIEDNAME(1, "Door"),
        doorTypeId,
        carDoorAttributes,
        nullptr,
        nullptr
    );
    UA_Server_addReference(
        _server,
        carDoorId,
        UA_NODEID_NUMERIC(0, UA_NS0ID_HASMODELLINGRULE),
        UA_EXPANDEDNODEID_NUMERIC(0, UA_NS0ID_MODELLINGRULE_MANDATORY),
        true
    );

    UA_NodeId carDoorColorId = UA_NODEID_NUMERIC(1, 6003);
    UA_VariableAttributes carDoorColorAttributes;
    UA_VariableAttributes_init(&carDoorColorAttributes);
    carDoorColorAttributes.displayName = UA_LOCALIZEDTEXT("en_US", "Color");
    UA_Variant_setScalar(&variant, &value, &UA_TYPES[UA_TYPES_INT32]);
    carDoorColorAttributes.value = variant;
    carDoorColorAttributes.dataType = UA_NODEID_NUMERIC(0, UA_NS0ID_INT32);
    carDoorColorAttributes.valueRank = -1;
    UA_Server_addVariableNode(
        _server,
        carDoorColorId,
        carDoorId,
        UA_NODEID_NUMERIC(0, UA_NS0ID_HASPROPERTY),
        UA_QUALIFIEDNAME(1, "Color"),
        UA_NODEID_NUMERIC(0, UA_NS0ID_PROPERTYTYPE),
        carDoorColorAttributes,
        nullptr,
        nullptr
    );
    UA_Server_addReference(
        _server,
        carDoorColorId,
        UA_NODEID_NUMERIC(0, UA_NS0ID_HASMODELLINGRULE),
        UA_EXPANDEDNODEID_NUMERIC(0, UA_NS0ID_MODELLINGRULE_MANDATORY),
        true
    );

    UA_NodeId* volkswagon;
    UA_ObjectAttributes volkswagonAttributes;
    UA_ObjectAttributes_init(&volkswagonAttributes);
    volkswagonAttributes.displayName = UA_LOCALIZEDTEXT("en_US", "Volkswagon");
    UA_Server_addObjectNode(
        _server,
        UA_NODEID_NULL,
        UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
        UA_NODEID_NUMERIC(0, UA_NS0ID_HASCOMPONENT),
        UA_QUALIFIEDNAME(1, "Volkswagon"),
        carTypeId,
        volkswagonAttributes,
        nullptr,
        volkswagon
    );

syvaren, 2017-05-10:
@jpfr Do I need to provide more information or can I help you debug this?

jpfr, 2017-05-11:
Hello @syvaren ,

on the current 0.2 branch, your code generates the following information model:

[image]

The door and the door color are automatically generated for the volkswagen object.
Which behavior did you expect there?

syvaren, 2017-05-12:
@jpfr This is expected behavior, and it does work on the 0.2 branch.  Is there some reason this doesn't work on the master branch?  I am working on master so I can have access to the session id in the method callback.

Stasik0, 2017-05-16:
@syvaren did you look into the ticket regarding method call context? https://github.com/open62541/open62541/issues/346

syvaren, 2017-05-16:
@Stasik0 I posted my usecase for the method call context, if that helps resolve that issue.  I am compiling against the master branch which includes that feature currently, but have not really had a chance to use it if it is not fully functional.

This particular issue is about having trouble with getting mandatory children to instantiate on master.  I see that a commit was made **instantiate non-mandatory children that already exist** (6b4947ad3331766989690d769817dee4edf2c4ed), not sure if that was intended to fix this, but it is still not working for me.  This does work on the 0.2 branch, which was recently merged to master.

[image]

Stasik0, 2017-05-16:
> This particular issue is about having trouble with getting mandatory children to instantiate on master. I see that a commit was made instantiate non-mandatory children that already exist (6b4947a), not sure if that was intended to fix this, but it is still not working for me. This does work on the 0.2 branch, which was recently merged to master.

@jpfr didnt we have a unit test for mandatory children?

XB32Z, 2017-07-06:
Hi, 

I'm not completely sure that's related but I was trying to model some complex objects too (for example following the OPC UA DI companion standard) and the objects appear without the main features of that standard. It's weird as the Types are perfectly fine. 
Did you tried to model objects following that standard?

Best,

jpfr, 2017-07-06:
Did you mark the members mandatory?
Then they should be instantiated just fine.
As described here: https://open62541.org/doc/0.2/tutorial_server_object.html

XB32Z, 2017-07-06:
Hi, 
I used the XML code generator from a model I made with UAModeler. I will try to build it programatically.

XB32Z, 2017-07-07:
Hi, 
I was actually wrong, the XML importer works quite well. However instanciating complex types produces errors. Additionally the XML importer call an instanciation which produce new nodes, duplicating those from the importer. In a general way, instanciation from a XML imported node set create lots of double (or triple) nodes and misses some others.

syvaren, 2017-09-13:
@Pro @jpfr 

This issue appears to be fixed in the latest master (0.3).  I am not sure which commit addressed this, but I am not missing any children today. :)

closed 2017-09-14 by Pro

syvaren, 2017-10-30:
@Pro 

I'm having trouble with this again.

This is the definition of my model, the child variables are mandatory.
[image]

Here is the instance of the parent, the object level is present, but the object's children are not.
[image]

syvaren, 2017-10-31:
@jpfr @Pro 

I've been messing around with this today, and still cannot get the children to populate.  Any ideas of what I can do to resolve this issue?