open62541/open62541 #1032: Read nodes folder of eWON

arnauddm, 2017-05-10:
Hi,

I use open62541 to read all tags of my eWON (gateway between Siemens and Server on Linux CentOS7) but when I list all nodes of my eWON, the library do what I want.

But my problem is that : all of my more important tags are in the folder "ewontags" (which is listed previously).

My question is : how can I list the attributes of this folder... ?

(I use the latest version of this library.)

FlorianPalm, 2017-05-10:
Hey,
basically, you should use a combination of Browse and Read services.

arnauddm, 2017-05-11:
How can I know if my readed value is a Float or an Int32 / 64... ?

FlorianPalm, 2017-05-11:
I am not sure if I understand your problem. If you read the value attribute of a VariableNode the datatype also available. You can check the datatype with (Scalar):
``C
UA_Variant_hasScalarType(const UA_Variant *v, const UA_DataType *type)
``

arnauddm, 2017-05-11:
Hum..
It is possible to get the type of the attribute before reading the attribute ?

closed 2017-05-11 by arnauddm

jpfr, 2017-05-11:
You can get the variable type before reading the value.
Every variable node has a `datatype` attribute that can be read separately.

However, the `datatype` attribute is a constraint.
For example. when the `datatype` attribute is a `Number`, the type of the value can be int, double, etc.