open62541/open62541 #1033: UA_ENABLE_MULTITHREADING missing header

aleks-f, 2017-05-10:
Description
===========

```
cmake -DUA_ENABLE_AMALGAMATION=true -DUA_ENABLE_MULTITHREADING=true ..
make
...
open62541.c:4012:19: fatal error: urcu.h: No such file or directory
```

Background Information / Reproduction Steps
===========================================

As describe in the example above.

Checklist
=========
Please provide the following information:

 - [x] open62541 Version (release number or git tag): 0.2-rc2
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: ubuntu 16.04
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [ ] Critical issue

Pro, 2017-05-10:
Can you try the most recent version of the 0.2 branch and see if it is fixed there?
rc2 is quite old

jpfr, 2017-05-10:
Hey @aleks-f ,

you need to install the liburcu library.
See here: https://open62541.org/doc/current/building.html

Please note that multithreading is still marked as experimental.

@Pro Well, 5 months is not so old... 😄 
It's the most recent release candidate. If we did an release candidate, mutlithreading worked on travis.

aleks-f, 2017-05-11:
@jpfr thanks, that worked.
not important for me, but on ubuntu 17.04:
```
$ sudo apt-get install sphinx graphviz
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package sphinx
```
and another question: if I generate code without `-DUA_ENABLE_MULTITHREADING`, is it safe/supported to wrap individual servers in my own threads (listening on different ports, of course)

jpfr, 2017-05-15:
Thanks, the package name was updated.

If you compile without UA_ENABLE_MULTITHREADING, you can run multiple servers at the same time.
However, you should avoid multiple threads that operate on the same server.
So every thread should have its own server(s).

closed 2017-05-15 by jpfr