open62541/open62541 #1034: open62541.dll

ArnaudMN, 2017-05-11:
Hi,

I would like to use open62541, but not in C/C++. I think (but i'm not use) it's possible to create a dll (open62541.dll) and use it with other language like Delphi ?
Is it possible to add compile version of the dll in github ?

Thank you

Stasik0, 2017-05-11:
you can have binaries from appveyor builds: https://ci.appveyor.com/project/open62541/
Pick a green for for a needed branch, e.g. 0.2.: https://ci.appveyor.com/project/open62541/open62541/build/2866 
Pick the right environment, e.g. 32 bit MSVS and click on "artifacts": https://ci.appveyor.com/project/open62541/open62541/build/2866/job/qk6yahbuanhcldil/artifacts

closed 2017-05-11 by Stasik0

ArnaudMN, 2017-05-11:
Thank you for your answer. But the first link ( https://ci.appveyor.com/project/open62541/) sayf "not found".
When I download the "artifacs", I have no ".dll" file, so I can't use it with my program.

jpfr, 2017-05-11:
- Take https://ci.appveyor.com/project/open62541/open62541/branch/0.2
- Select a compiler
- Download artefacts

ArnaudMN, 2017-05-11:
Yes I understand .... but there are no "open62541.dll" in the zip file.
I just want a shared library (like explain at [https://open62541.org/doc/current/building.html#building-with-cmake-on-windows](https://open62541.org/doc/current/building.html#building-with-cmake-on-windows)