open62541/open62541 #1035: Discovery Unit Tests fail on master

jpfr, 2017-05-11:
Description
===========

The discovery unit tests fails on the master branch.

https://paste2.org/j2N5bhZe

@Pro Do you have an intuition where this comes from?

Background Information / Reproduction Steps
===========================================

See here for the kinds of information we need to reproduce the bug and how to obtain it:
https://github.com/open62541/open62541/wiki/Writing-Good-Issue-Reports

Checklist
=========
Please provide the following information:

 - [ x] open62541 Version (release number or git tag): master
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ x] Operating system: Linux
 - [ x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Pro, 2017-05-11:
The reported error means `BadServiceUnsupported` so it looks like the code wasn't compiled with the `UA_ENABLE_DISCOVERY_MULTICAST` flag?

jpfr, 2017-05-11:
Can we surround the unit tests that require `UA_ENABLE_DISCOVERY_MULTICAST` with the same #if macro?
It's not good to have some unit tests fail on travis. That leads to other regressions not being noticed.

Pro, 2017-05-11:
Yes, that would be a good option, but currently `UA_ENABLE_DISCOVERY_MULTICAST` is always enabled for unit tests on travis:
https://github.com/open62541/open62541/blob/master/tools/travis/travis_linux_script.sh#L167

So that shouldn't be an issue here?

jpfr, 2017-05-15:
Unit tests should never fail.
Independently of whether they are run on travis or locally.

Pro, 2017-06-07:
Fixed in https://github.com/open62541/open62541/commit/cd665efcf34d5570731f3617e25f9c02d58172e4

commit a05b620c (2017-05-16): Disable unit tests if define is not set. Fix #1035
commit cd665efc (2017-05-18): Disable unit tests if define is not set. Fix #1035
PR #1036 (2017-05-19): Only enable multicast unit tests if multicast is enabled
@jpfr does this fix #1035 for you?
closed 2017-06-07 by Pro