open62541/open62541 #1038: Value changed,but  the Subscription handle doesn't work

yealfor, 2017-05-16:
Description
===========
I instanced a server,and change the value of one node. Then I  Subscribe that node, but I can't receive the handler function seem not working, where I did wrong?Help! 

Background Information / Reproduction Steps
===========================================

See here for the kinds of information we need to reproduce the bug and how to obtain it:
https://github.com/open62541/open62541/wiki/Writing-Good-Issue-Reports

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Pro, 2017-05-16:
Can you post a minimum working example code which shows your Problem (https://stackoverflow.com/help/mcve)?
Which version of open62541 do you use?

Pro, 2017-05-16:
https://github.com/open62541/open62541/wiki/Writing-Good-Issue-Reports

yealfor, 2017-05-16:
The code is same as the example in "client.c",but the Node(1,1870) is a new node which I creaked in Namespace through add node xml and recompile the dll.
	/* Create a subscription */
    UA_UInt32 subId = 0;
    UA_Client_Subscriptions_new(client, UA_SubscriptionSettings_standard, &subId);
    if(subId)
        printf("Create subscription succeeded, id %u\n", subId);
	/* Add a MonitoredItem */
    UA_NodeId monitorThis = UA_NODEID_NUMERIC(1, 1870);
    UA_UInt32 monId = 0;
    UA_Client_Subscriptions_addMonitoredItem(client, subId, monitorThis, UA_ATTRIBUTEID_VALUE,
                                             &handler_TheAnswerChanged, NULL, &monId);
    if (monId)
        printf("Monitoring 'the.answer', id %u\n", subId);
    /* The first publish request should return the initial value of the variable */
    UA_Client_Subscriptions_manuallySendPublishRequest(client);
	
And I make a timer to write the value of that node with a random number ,like this:
	/* Write node attribute */
    int value=qrand();
    UA_WriteRequest wReq;
    UA_WriteRequest_init(&wReq);
    wReq.nodesToWrite = UA_WriteValue_new();
    wReq.nodesToWriteSize = 1;
    wReq.nodesToWrite[0].nodeId = UA_NODEID_NUMERIC(1, 1870);
    wReq.nodesToWrite[0].attributeId = UA_ATTRIBUTEID_VALUE;
    wReq.nodesToWrite[0].value.hasValue = true;
    wReq.nodesToWrite[0].value.value.type = &UA_TYPES[UA_TYPES_INT32];
    wReq.nodesToWrite[0].value.value.storageType = UA_VARIANT_DATA_NODELETE; /* do not free the integer on deletion */
    wReq.nodesToWrite[0].value.value.data = &value;
    UA_WriteResponse wResp = UA_Client_Service_write(client, wReq);
    if(wResp.responseHeader.serviceResult == UA_STATUSCODE_GOOD)
        qDebug()<<QString("Writing value:%1").arg(value);
    UA_WriteRequest_deleteMembers(&wReq);
    UA_WriteResponse_deleteMembers(&wResp);
	
Then, the problem is the handler function is not display the new value of that node:
	static void
handler_TheAnswerChanged(UA_UInt32 monId, UA_DataValue *value, void *context) {
    printf("The Answer has changed!\n");
}

Besides,I found other user solve this problem through call  "UA_Client_Subscriptions_manuallySendPublishRequest(client)" in a loop,but couldn't the subscription call the handler function automatically?

yealfor, 2017-05-16:
And I am using  version  open62541-0.2-rc2

Pro, 2017-05-16:
Can you try your same code with the most recent version of 0.2 branch:
https://github.com/open62541/open62541/tree/0.2
This includes many bugfixes since the rc2 release

jpfr, 2017-05-16:
Do you see the debug log of
```c
if(wResp.responseHeader.serviceResult == UA_STATUSCODE_GOOD)
        qDebug()<<QString("Writing value:%1").arg(value);
```

Maybe the value is never written due to a type mismatch.
I am **very sure** that subscriptions are working in 0.2-rc2 in general.

jpfr, 2017-05-16:
Also, have a look at `wResp.results[0]` to get the status of the first write operation.

yealfor, 2017-05-16:
Yes, I use Qt to program,the debug log will appear aright. And I can see that the value has been written in address space using UAexpert

yealfor, 2017-05-16:
If I call "UA_Client_Subscriptions_manuallySendPublishRequest(client)" in a loop,I can obtain the right value which is written

yealfor, 2017-05-16:
Perhaps, could you give me a  example code which can work correctly.

Stasik0, 2017-05-16:
this is the outstanding eventing in client feature, that is free for grabs https://github.com/open62541/open62541/issues/905

jpfr, 2017-05-16:
> this is the outstanding eventing in client feature,

Yes. Async services are a big big requirement for the client.

Currently, you can send a publish request every 200ms. 5 packets/sec is very little traffic.
And you get all the notifications from that period. Since the MonitoredItems can use a much smaller sampling interval internally. So you get *nearly all* the benefits of push notifications.

Still, async service calls are a big big requirement.

yealfor, 2017-05-17:
Ok, It is seem that manually sending request  is the most appropriate method currently.....
Thank you very much!

closed 2017-05-17 by jpfr