open62541/open62541 #1041: Adding empty variable node results in

Pro, 2017-05-16:
(On current master branch)

If a variable node is added without any value, the call fails with `UA_STATUSCODE_BADTYPEMISMATCH` and `Only Variables with data type BaseDataType may contain a null (empty) value`

The following example demonstrates the problem (from https://github.com/open62541/open62541/blob/master/tests/check_client_highlevel.c#L331):
```
            UA_VariableAttributes attr;
            UA_VariableAttributes_init(&attr);
            attr.description = UA_LOCALIZEDTEXT("en_US", "Array");
            attr.displayName = UA_LOCALIZEDTEXT("en_US", "Array");
			attr.accessLevel = UA_ACCESSLEVELMASK_READ | UA_ACCESSLEVELMASK_WRITE;
			attr.writeMask = UA_WRITEMASK_ARRRAYDIMENSIONS;

            /*UA_Int32 values[2];
            values[1] = 10;
            values[2] = 20;

            UA_Variant_setArray(&attr.value, values, 2, &UA_TYPES[UA_TYPES_INT32]);*/
            attr.dataType = UA_TYPES[UA_TYPES_INT32].typeId;
            attr.valueRank = 1; /* array with one dimension */
            UA_UInt32 arrayDims[1] = {2};
            attr.arrayDimensions = arrayDims;
            attr.arrayDimensionsSize = 1;

            retval = UA_Client_addVariableNode(client, UA_NODEID_NULL,
                                               unitTestNodeId,
                                               UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES),
                                               UA_QUALIFIEDNAME(1, "Array"),
                                               UA_NODEID_NULL, attr, &nodeArrayId);
            ck_assert_uint_eq(retval, UA_STATUSCODE_GOOD);
```

I think this is related to the line https://github.com/open62541/open62541/blob/master/src/server/ua_services_nodemanagement.c#L534 which then causes the following to fail:

- Service_AddNode_finish
- typeCheckVariableNode (https://github.com/open62541/open62541/blob/master/src/server/ua_services_nodemanagement.c#L762)
- typeCheckVariableNodeWithValue (https://github.com/open62541/open62541/blob/master/src/server/ua_services_nodemanagement.c#L231)
- typeCheckValue (https://github.com/open62541/open62541/blob/master/src/server/ua_services_nodemanagement.c#L140)
- ERROR: `Only Variables with data type BaseDataType may contain a null (empty) value` (https://github.com/open62541/open62541/blob/master/src/server/ua_services_attribute.c#L240)

If I create the `Array` variable, then I'm not able to change the array dimensions in a later step (https://github.com/open62541/open62541/blob/master/tests/check_client_highlevel.c#L417)

Or should we change the unit test?

jpfr, 2017-05-16:
You should be able to change the array dimensions in that case.
But the ordering is important such that the variable is never inconsistent.

- Set no arraydimensions at all.
- Change the value to fit the new array dimensions
- Set the new array dimensions

Pro, 2017-05-16:
Ok, I tried now with the following code, which does not set any arraydimensions when creating the node:

```
            UA_VariableAttributes attr;
            UA_VariableAttributes_init(&attr);
            attr.description = UA_LOCALIZEDTEXT("en_US", "Array");
            attr.displayName = UA_LOCALIZEDTEXT("en_US", "Array");
			attr.accessLevel = UA_ACCESSLEVELMASK_READ | UA_ACCESSLEVELMASK_WRITE;
			attr.writeMask = UA_WRITEMASK_ARRRAYDIMENSIONS;

            /*UA_Int32 values[2];
            values[1] = 10;
            values[2] = 20;

            UA_Variant_setArray(&attr.value, values, 2, &UA_TYPES[UA_TYPES_INT32]);*/
            attr.dataType = UA_TYPES[UA_TYPES_INT32].typeId;

            retval = UA_Client_addVariableNode(client, UA_NODEID_NULL,
                                               unitTestNodeId,
                                               UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES),
                                               UA_QUALIFIEDNAME(1, "Array"),
                                               UA_NODEID_NULL, attr, &nodeArrayId);
            ck_assert_uint_eq(retval, UA_STATUSCODE_GOOD);
```

The problem is still there:
ERROR: `Only Variables with data type BaseDataType may contain a null (empty) value` (https://github.com/open62541/open62541/blob/master/src/server/ua_services_attribute.c#L240)

You can also check the branch in here:
https://github.com/open62541/open62541/tree/testing/fix_tests

When this is fixed and the branch merged, all the unit tests should again be successful :+1:

commit c60f7777 (2017-08-23): Fix #1041: Only perform compatibility checks if there is a value type set

value.hasValue only indicates, if the value member is initialized,
but not if the value is also set. Only if value.value.type is not NULL
there is a value.
closed 2017-08-24 by Pro