open62541/open62541 #1042: Translation from UA_NodeId to UA_DataType

aleks-f, 2017-05-18:
Description
===========
I'm trying to understand how translation from [UA_NodeId](https://github.com/open62541/open62541/blob/v0.2/include/ua_types.h#L298) to [UA_DataType](https://github.com/open62541/open62541/blob/v0.2/include/ua_types.h#L768) using [UA_findDataType](https://github.com/open62541/open62541/blob/v0.2/src/ua_types.c#L31) works.

Background Information / Reproduction Steps
===========================================

Code:
```
include <iostream>
#include "open62541.h"
using namespace std;

void printDataType(const UA_NodeId& nIDType)
{
	const UA_DataType* pDT = UA_findDataType(&nIDType);
	if(pDT)
	{
		cout << "name: " << pDT->typeName << endl;
		cout << "memSize: " << pDT->memSize << endl;
		cout << "typeIndex: " << pDT->typeIndex << endl;
		cout << "membersSize: " << (int)pDT->membersSize << endl;
		cout << "builtin: " << pDT->builtin << endl;
		cout << "fixedSize: " << pDT->fixedSize << endl;
		cout << "overlayable: " << pDT->overlayable << endl;
		cout << "binaryEncodingId: " << pDT->binaryEncodingId << endl;
	}
}

int main()
{
	UA_NodeId valueNodeId = UA_NODEID_STRING_ALLOC(1, "the.answer");
	printDataType(valueNodeId);
	return 0;
}
```
Output:
```
name: Float
memSize: 4
typeIndex: 9
membersSize: 1
builtin: 1
fixedSize: 1
overlayable: 1
binaryEncodingId: 0
```
Question 1: If `UA_NodeId::identifier` is an union, why are types sought based on numeric value only, returning incorrect results for anything but integer values?

Question 2:  if [UA_NODEIDTYPE_NUMERIC](https://github.com/open62541/open62541/blob/v0.2/include/ua_types.h#L281) can be 1, 2 or 4 byte integer, why are enum members `1` and `2` not provided?

Checklist
=========
Please provide the following information:

 - [x] open62541 Version (release number or git tag): 0.2
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Ubuntu Linux 16.04
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2017-05-20:
> Question 1: Why are types sought based on the UA_NodeId::identifier::numeric, rather than UA_NodeId::identifierType?

We want the type lookup to be fast, since it has to be done for every request.
The standard-defined types all have a numeric node-nodeid. Did you encounter an actual datatype with a string nodeid or similar?
We can check the identifierType in the beginning and switch to a different / slow lookup mode to be complete..

> Question 2: if UA_NODEIDTYPE_NUMERIC can be 1, 2 or 4 byte integer, why are enum members 1 and 2 not provided?

It simplifies the user-facing API. 1- and 2-byte nodeids always have a numeric identifier. We transparently switch to the correct binary encoding in the serialisation layer. That depends for example on
- if the identifier fits into a 16bit integer
- if the namespace is zero

jpfr, 2017-05-20:
Question 1 is addresses on master in commit be215e3b1ff597f231f9f5d3e63ebcc483421070.
The standard data types all have a numeric identifier. We now make that additional check.

jpfr, 2017-05-20:
Please reopen if required.

closed 2017-05-20 by jpfr