open62541/open62541 #1043: Ua Expert gets 'BadSecurityModeInsufficent' during activatesession

pinkiecao, 2017-05-18:
Hello All,
    I develop a server with open62541. I want to make the server support the username/password authentication. So I set the config:

    UA_ServerConfig config = UA_ServerConfig_standard;
     config.enableAnonymousLogin = false;
     config.enableUsernamePasswordLogin = true;
     config.usernamePasswordLoginsSize = 1;
     UA_UsernamePasswordLogin logininfo;
     logininfo.username = UA_STRING("user1");
     logininfo.password = UA_STRING("password1");
     config.usernamePasswordLogins = &logininfo;

But UaExpert gets BadSecurityModeInsufficent error.  I think it is about securitypolicy. The 62541 manual says only 'None' policy is supported so far. Does anyone have used the authentication? Please tell me ,thanks a lot.

Stasik0, 2017-07-07:
did you try it with default server? https://github.com/open62541/open62541/blob/2cabd7c5d960dadcebc75c194daafc09cfa74ce2/plugins/ua_config_standard.c#L38

Stasik0, 2017-07-07:
[image]

Pro, 2017-08-13:
@pinkiecao I assume this is solved?
If not please reopen and post any comment. Thanks

closed 2017-08-13 by Pro

rashed043, 2017-12-04:
Hello all,
i do have the same problem "username/password authentication" and get the same the error  BadSecurityModeInsufficent. Could you please tell me how did you solve the problem. Thank a lots in advance..

rashed