open62541/open62541 #1044: UInteger inside Integer DataType

IzonFreak, 2017-05-18:
In the DataTypes folder the UInteger DataType and his children are inside the Integer DataType. Shouldn't it be at the same level as Integer, Double and Float? 

I checked other servers (Prosys Public Server) and the green book "OPC Unified Architecture" in page 335 and they all have the same structure with UInteger, Integer, Double and Float inheriting from Number Data Type directly. 

SDK Version 0.2.0, with UAExpert Client

[image]

Pro, 2017-05-18:
Thanks for the report!

It looks like this was changed between schema version 1.02 and 1.03:

The `NodeSet2.xml?`of 1.02 says `UInteger` isSubtype of `Integer` (https://opcfoundation.org/UA/schemas/1.02/Opc.Ua.NodeSet2.xml):

```
  <UADataType NodeId="i=26" BrowseName="Number" IsAbstract="true">
    <DisplayName>Number</DisplayName>
    <Description>Describes a value that can have any numeric DataType.</Description>
    <References>
      <Reference ReferenceType="HasSubtype" IsForward="false">i=24</Reference>
    </References>
  </UADataType>
  <UADataType NodeId="i=27" BrowseName="Integer" IsAbstract="true">
    <DisplayName>Integer</DisplayName>
    <Description>Describes a value that can have any integer DataType.</Description>
    <References>
      <Reference ReferenceType="HasSubtype" IsForward="false">i=26</Reference>
    </References>
  </UADataType>
  <UADataType NodeId="i=28" BrowseName="UInteger" IsAbstract="true">
    <DisplayName>UInteger</DisplayName>
    <Description>Describes a value that can have any unsigned integer DataType.</Description>
    <References>
      <Reference ReferenceType="HasSubtype" IsForward="false">i=27</Reference>
    </References>
  </UADataType>
```

whereas 1.03 (https://opcfoundation.org/UA/schemas/1.03/Opc.Ua.NodeSet2.xml)  says `UInteger` isSubtype of `Number`

```
  <UADataType NodeId="i=26" BrowseName="Number" IsAbstract="true">
    <DisplayName>Number</DisplayName>
    <Description>Describes a value that can have any numeric DataType.</Description>
    <References>
      <Reference ReferenceType="HasSubtype" IsForward="false">i=24</Reference>
    </References>
  </UADataType>
  <UADataType NodeId="i=27" BrowseName="Integer" IsAbstract="true">
    <DisplayName>Integer</DisplayName>
    <Description>Describes a value that can have any integer DataType.</Description>
    <References>
      <Reference ReferenceType="HasSubtype" IsForward="false">i=26</Reference>
    </References>
  </UADataType>
  <UADataType NodeId="i=28" BrowseName="UInteger" IsAbstract="true">
    <DisplayName>UInteger</DisplayName>
    <Description>Describes a value that can have any unsigned integer DataType.</Description>
    <References>
      <Reference ReferenceType="HasSubtype" IsForward="false">i=26</Reference>
    </References>
  </UADataType>
```

So I think we should update our NodeSet from 1.02 to 1.03 definitely in master, but also as a bugfix release for 0.2? @jpfr?

jpfr, 2017-05-18:
We can change it to the 1.03 version on 0.2 and master.
But I would not make a bugfix release specifically for this issue.

When we are already on it, does the diff between the 1.02 and 1.03 schema show any other significant changes in the ns0 nodes we bootstrap by hand?

Pro, 2017-05-18:
I uploaded the diff here:
https://www.dropbox.com/s/q10wljgv8rwpawl/nodeset_diff.htm?dl=1

There are some other changes which may affect our manually bootstrap nodes. Maybe there is an easy way on creating an automatic unit test which reads the xml and does the corresponding checking of these bootstrap nodes.

Btw: with bugfix release I didn't meant a specific release, but include it into the upcoming bugfix release ;)

jpfr, 2017-06-06:
> Maybe there is an easy way on creating an automatic unit test which reads the xml and does the corresponding checking of these bootstrap nodes.

Relevant xkcd: https://xkcd.com/1205/
How does this compare this to other open points that require dev capacity?

Also with automatic testing, we can just manually make the change on master and close the issue.

Pro, 2017-06-06:
I updated the schema now manually without unit tests.
That's a nice to have feature and probably not worth the time according to xkcd

PR #1067 (2017-06-06): Update to schema version 1.03
Files downloaded from https://opcfoundation.org/UA/schemas/1.03/

Fix #1044
closed 2017-06-06 by Pro