open62541/open62541 #1047: Cannot add array variables to server

CPalas, 2017-05-22:
--------------------------------------------------------------------------------

Description
===========
I need to manage some array variables. The C program in attachment is built just copying the code from the tutorial at
https://open62541.org/doc/current/tutorial_server_variabletype.html
after successful compilation it runs, but no 2Dpoint variable or any 2D type appears in the addressspace browsed by UA expert or any client I can program.

Thank you very much.
Best

Background Information / Reproduction Steps
===========================================
I am working with a mac on which i recently installed OPC UA 62451 by following the instructions at https://open62541.org/doc/current/building.html (Building on OS X)
The program compiles with 
gcc -std=c99 open62541.c addVariable2Dserver.c -o addVariable2Dserver
with minor warnings  and the server can be launched with
./addVariable2Dserver
I do not see any array variable, neither with UA Expert or launching some client such as the one from the Open 62541examples. The compilation warnings i get do not prevent more complex server programs I programmed to run and create scalar variables.

Checklist
=========
Please provide the following information:

 - [ x] open62541 Version (release number or git tag):
The open62541.c and .h files I downloaded are marked * Git-Revision: 0.2-rc2

 - [X] Other OPC UA SDKs used (client or server): 
I am using UA Expert to browse my server
 - [X ] Operating system:
OS X El capitan 10.11.6

 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [X] Self-contained code example attached
[addVariable2Dserver.txt](https://github.com/open62541/open62541/files/1019017/addVariable2Dserver.txt)
 - [ ] Critical issue

Pro, 2017-05-22:
Thanks for your detailed issue report!
I just ran your code 1:1 and I can see the node here on linux:

[image]

I'm using the current master branch. Can you try the latest 0.2 release https://github.com/open62541/open62541/releases/tag/v0.2. If that doesn't work, please try master and report your findings.

CPalas, 2017-05-22:
It works. 
I just copied and pasted into my working directory the two new files
open62541.h and open62541.c 
from  https://github.com/open62541/open62541/releases/tag/v0.2
Thank you very much indeed.
BR

closed 2017-05-22 by CPalas