open62541/open62541 #1048: Adding variable node fails when namespace index != 1

aleks-f, 2017-05-23:
Description
===========

Trying to add a variable node with namespace index different than `1` fails with
```
BadNodeIdInvalid: The syntax of the node id is not valid.
```

Background Information / Reproduction Steps
===========================================

Using the [server example](https://github.com/open62541/open62541/blob/v0.2/examples/server.cpp#L46), if I try to create a variable node with namespace index != 1, it will fail with the above error.

Am I missing something here?

Checklist
=========
Please provide the following information:

 - [x] open62541 Version (release number or git tag): 0.2
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: ubuntu 16.04
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Stasik0, 2017-05-23:
You need to call UA_Server_addNamespace

aleks-f, 2017-05-23:
yes, that was it, thanks.

closed 2017-05-23 by aleks-f