open62541/open62541 #1050: Segmentation fault after running 3 days with above 1000 real time updating nodes

JinGuoGe, 2017-05-24:
Description
===========
Hi, 
I am using single file sdk - open62541.c/open62541.h  for opcserver.
I created about 1000 string value nodes. And internally, update string value every second.
On the other hand,  sent monitoring request from opcuaclient which is developed by eclipse-milo opc-ua sdk.
After running 3 days, it get crashed.
I am running in device which is installed debian system and have 512Mbyte memory.

Background Information / Reproduction Steps
===========================================
Following url is server-example code.
https://paste2.org/NOXyPB6X
And for opc ua client, used eclipse-milo sdk and sent monitor request for 1000 nodes.

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag): 3a137608dc
 - [ ] Other OPC UA SDKs used (client or server):  
      eclipse milo sdk for client
 - [ ] Operating system: 
      debian
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached

      Following url is crashed log of gdb.
      https://paste2.org/KneaeV68

 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2017-05-24:
Hello @JinGuoGe ,

The service `UA_Server_writeValue` internally makes a copy of the value.
At this code line
```c
	UA_Variant_setScalarCopy(&node_value, &ua_str_value, &UA_TYPES[UA_TYPES_STRING]);
	ret_status = UA_Server_writeValue(g_server, data_node_id, node_value);
        return ret_status;
```
you could try to just use `UA_Variant_setScalar` or to call `UA_Variant_deleteMembers(&node_value)` afterwards.

On Debian, you can use the tool valgrind to check for memleaks and other memory errors.
These should show after a short time already.
So you can close the server after a short time and see whether valgrind reports errors.

open62541 is valgrind-clean. It could be that you run out of RAM or otherwise overwrite some memory.
Does the same error occur with a "valgrind-clean" version of you application?
Can you then paste the valgrind output? It will give us even more information.

JinGuoGe, 2017-05-24:
Hi, thanks for your response.
I saw  memory-leak by valgrind. You pointed out problem correctly.
Valgrind also pointed out same code line.
But i don't understand why encoding length is 1221581600 in above crashed log. It can be because of this? If so,  i will close this issue.

jpfr, 2017-05-24:
>  But i don't understand why encoding length is 1221581600 in above crashed log. It can be because of this?

Maybe. Anyway, it makes sense to have no memleaks in a long-running app.
If this error goes away as well, all the better.

JinGuoGe, 2017-05-24:
Ok, then i will test more for about 3 days. Thanks for your solution.

JinGuoGe, 2017-05-30:
Hi, After testing 5 days, also crashed. 
And i also checked with Valgrind.
`6665 errors in context 27 of 30:
==9321== Invalid read of size 2
==9321==    at 0x4C30908: memcpy@@GLIBC_2.14 (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==9321==    by 0x4E55EA1: UInt32_encodeBinary (open62541.c:6019)
==9321==    by 0x4E5776C: Variant_encodeBinary (open62541.c:6795)
==9321==    by 0x4E57C2A: DataValue_encodeBinary (open62541.c:6910)
==9321==    by 0x4E5838D: UA_encodeBinaryInternal (open62541.c:7091)
==9321==    by 0x4E584DE: UA_encodeBinary (open62541.c:7128)
==9321==    by 0x4E75773: detectValueChange (open62541.c:23609)
==9321==    by 0x4E75BFF: UA_MoniteredItem_SampleCallback (open62541.c:23664)
==9321==    by 0x4E6485E: processJob (open62541.c:17900)
==9321==    by 0x4E64B83: processRepeatedJobs (open62541.c:18092)
==9321==    by 0x4E652D1: UA_Server_run_iterate (open62541.c:18435)
==9321==    by 0x4E655F5: UA_Server_run (open62541.c:18537)
==9321==  Address 0x5b72410 is 0 bytes inside a block of size 4 free'd
==9321==    at 0x4C2CE10: free (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==9321==    by 0x4E55AD5: UA_Array_delete (open62541.c:5781)
==9321==    by 0x4E5402B: Variant_deletemembers (open62541.c:5121)
==9321==    by 0x4E55104: DataValue_deleteMembers (open62541.c:5489)
==9321==    by 0x4E557C4: deleteMembers_noInit (open62541.c:5698)
==9321==    by 0x4E5585F: UA_deleteMembers (open62541.c:5712)
==9321==    by 0x4E511AE: UA_DataValue_deleteMembers (open62541.h:5731)
==9321==    by 0x4E6A7EF: writeValueAttributeWithoutRange (open62541.c:20439)
==9321==    by 0x4E6AC67: writeValueAttribute (open62541.c:20515)
==9321==    by 0x4E6C1D7: CopyAttributeIntoNode (open62541.c:21020)
==9321==    by 0x4E6476F: UA_Server_editNode (open62541.c:17815)
==9321==    by 0x4E6CAA1: UA_Server_write (open62541.c:21136)`

jpfr, 2017-05-31:
Hello @JinGuoGe ,

based on the paste, the error comes from trying to encode a value that was previously deleted.
Are you interacting with the server from multiple threads?

If not, we'd have to create a reproducable test case / unit test based on the information you can give us.

JinGuoGe, 2017-06-02:
Hi
There are two threads- one is opcserver running thread, and the other is writing to opcserver nodes with updated values.
Is this structure problem?

jpfr, 2017-06-03:
That's what I thought. You are encoding a value that does no longer exist.
You need to protect the server with a mutex, so no two threads can access it at the same time.

We have a build option to make open62541 multithreading-safe.
But the feature is experimental and not recommended for production use as of now.
Multithreading will be in a better shape for 0.3 if PR #900 lands in time for it.

Please reopen if required.

closed 2017-06-03 by jpfr

JinGuoGe, 2017-06-07:
Hi,
I will write here my test opcserver  code. it  still crashes after 3 days running.
why does it need multihreading-safe? Could you explain in detail?
https://paste2.org/nEVbAXmV