open62541/open62541 #1051: Binary encoding/chunking bug

bostjanv, 2017-05-25:
Description
===========

Possible bug in binary encoding/chunking code.

Background Information / Reproduction Steps
===========================================

We are developing a specific application with soft-realtime constraints. We rely on open62541 stack for communication and data transfer on both server and client side. A couple of weeks ago while performing extensive long-running tests we started to see random and rare timeouts/disconnections. Since the application code is rather complex our initial guess was that we have some timing/synchronization issues in our code. But after some extensive debugging sessions it turned out the we (very likely) triggered a bug in the open62541 implementation of binary encoding/chunking.

After some further investigations I found out that the bug shows itself in the case where there is not enough space in the current chunk to encode the length of the array. The problematic function is Array_encodeBinary, explicitly this block of code:
`/* Encode the array length */
    UA_StatusCode retval = Int32_encodeBinary(&signed_length);
    if(retval != UA_STATUSCODE_GOOD || length == 0)
        return retval;`

I am not excluding the possibility that a similar problem exists in some other part of the code as well.

I tried to fix the issue by placing another exchangeBuffer call at the strategic place by this resulted in some random decoding errors on the client side. I suppose I am still missing something and my fix was not correct.

I've prepared two unit tests that trigger the bug. See https://github.com/bostjanv/open62541/commit/a7bee9ac884696f66b09f81401f4c0acf325532d

Disclaimer: Please not that since I'm not intimately familiar with the code my reasoning might be wrong.

Checklist
=========

 - [x] open62541 Version (release number or git tag): 0.2, master
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [x] Critical issue

jpfr, 2017-05-25:
Thanks for a perfect bug report. The fix lives in #1052.
It will be merged as soon as travis gives a green light.

The encoding throws a specific error when the end of the buffer is reached.
This is detected, the buffer replaced, and the encoding is restarted at the last "checkpoint" where a complete datatype was encoded.

There was a rare code path where the buffer replacement was not done.
For triggering this, the 4-byte length had to exactly cross the buffer boundary.
Then, encoding was aborted with an error code.

jpfr, 2017-05-25:
@bostjanv Can you confirm the CLA in PR #1052?
With this, we can include your unit tests to the main code base.

bostjanv, 2017-05-25:
Signed

bostjanv, 2017-05-26:
Actually your fix is exactly the same as was mine that I am referring to in the above description. Unfortunately, I think the fix is not correct, since after applying the fix I was getting decoding errors at the client side. To be completely sure, I've prepared another example https://github.com/bostjanv/open62541/commit/dabfb61a5d770f2b7a91b6ba4a404d3b4c19a9e5 which demonstrates my findings. Running the example with valgrind shows also some errors which further supports my claim.

jpfr, 2017-05-27:
Fixed via #1052.
@bostjanv Can you confirm?

bostjanv, 2017-05-29:
Thanks! This was far from trivial :-) I think this finally resolves it.

jpfr, 2017-05-29:
Cool, glad to heat that.

closed 2017-05-29 by jpfr