open62541/open62541 #1053: How to publish and retrieve DateTime

aleks-f, 2017-05-25:
Description
===========
Adding DateTime variable node to server works fine. I can retrieve it using client without any errors reported, but i do not get what I've pubished on the server side but the "beginning of time" (ie. 01/01/1601 ...)

Background Information / Reproduction Steps
===========================================
server:
```
UA_VariableAttributes attr;
UA_VariableAttributes_init(&attr);
UA_DateTime dt = UA_DateTime_now();
UA_Variant_setScalarCopy(&attr.value, &dt, &UA_TYPES[UA_TYPES_DATETIME]);
attr.description = UA_LOCALIZEDTEXT_ALLOC("en_US","the date");
attr.displayName = UA_LOCALIZEDTEXT_ALLOC("en_US","the date");
UA_NodeId nodeId = UA_NODEID_NUMERIC(1, 1);
UA_QualifiedName name = UA_QUALIFIEDNAME_ALLOC(1, "the date");
UA_NodeId parentNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER);
UA_NodeId parentReferenceNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES);
UA_Server_addVariableNode(server, nodeId, parentNodeId,
                              parentReferenceNodeId, name,
                              UA_NODEID_NULL, attr, NULL, NULL);
```

client side:
```
UA_Variant val;
UA_Client_readValueAttribute(client, UA_NODEID_NUMERIC(1, 1), &val);
UA_DateTime dt = *(UA_DateTime*)val->data;
std::cout <<
UA_DateTime_toString(dt)
<< std::endl;
```

output:
```
01/01/1601 00:00:01.000.000.600
```

what am I doing wrong?

Checklist
=========
Please provide the following information:

 - [x] open62541 Version (release number or git tag): 0.2
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: ubuntu 16.04
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Pro, 2017-05-26:
I just tested your code in C and it works with:
```
    UA_String string_date = UA_DateTime_toString(dt);
    printf("string date is: %.*s\n", (int)string_date.length, string_date.data);
```

The only thing different is the `operator<<` which you are using. I do not recall that it is defined for UA_String.

You can try the following:
```
    UA_DateTime dt = *(UA_DateTime*)val->data;

    UA_String string_date = UA_DateTime_toString(dt);
    std::cout << std::string((char*)string_date.data, string_date.length) << std::endl;
```

aleks-f, 2017-05-26:
no, that was just a copy/paste mistake. I was retrieving the wrong value from server in the code - it does indeed work as advertised, thanks for looking into it.

closed 2017-05-26 by aleks-f