open62541/open62541 #1055: Bad validation in copySubString() in ua_types.c

jonasgreen88, 2017-05-29:
Description
===========
I'm trying to define the DataTypeSystem "OPC Binary" and a DataTypeDictionary node as a component of the DataTypeSystem. The DataTypeDictionary node is a Variable node with data type ByteString. UA Expert sends a read request to this node once a connection to the server was setup. In this read access UA Expert includes a range to read (index 0 to index 0x007FFFFF). The request resulted in an Access violation exception (Visual Studio project) in `UA_Array_delete()`, ua_types.c. I tracked the cause of this issue to the second if-statement in the function `copySubString()` in ua_types.c. The if statement checks if the max dimension (in this case 0x007FFFFF) was equal or larger than the length of the actual ByteString (which in this case was true). As the condition was true the status code `UA_STATUSCODE_BADINDEXRANGENODATA` was returned. The exception happen while trying to clean up the destination buffer after the copy has failed.

The actual if statement causing the problem:
`if(dim->max >= src->length)
return UA_STATUSCODE_BADINDEXRANGENODATA;`

When reading OPC UA Part 4, chapter 7.22 NumericRange it says:
> When reading a value, the indexes may not specify a range that is within the bounds of the array. The Server shall return a partial result if some elements exist within the range.

In my opinion the implementation does not conform to this statement. If a part of the range is within the  bounds of the array this part shall be returned in a success response to my understanding. However, I'm kind of new to OPC UA so I may have missed something important here. What do you guys think about this?

Background Information / Reproduction Steps
===========================================
Use the attached application and start the server. Connect to the server with UA Expert 1.4.4.

Checklist
=========
Please provide the following information:

 - [x] open62541 Version (release number or git tag): be5af5786998c3ddfc9e36858dc86cff1368cfbd
 - [x] Other OPC UA SDKs used (client or server): UAExpert 1.4.4 as client
 - [x] Operating system: Windows 7, running the stack as server in Visual Studio.
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
[ua_log.txt](https://github.com/open62541/open62541/files/1036003/ua_log.txt)
 - [x] Wireshark network dump attached: Running both client and server on localhost
 - [x] Self-contained code example attached: Attached my application using the stack in Visual Studio (playing around with structured data types).
[appl.zip](https://github.com/open62541/open62541/files/1035988/appl.zip)
 - [ ] Critical issue

jpfr, 2017-05-29:
Right. We should change that to be spec-conform.
It will be enough to just adjust the returned length within copySubString.

I'm wondering why there is no test case covering this in the official testing tools....

jpfr, 2017-07-04:
Fixed via #1095.

Thanks for the wonderfully complete report.

Next time there is a bug involving a crash, please highlight that part.
So the issue will get more attention.

commit 089dd7aa (2017-07-04): return partial content of a range (fix #1055)
closed 2017-07-04 by jpfr