open62541/open62541 #1057: Namespace compiler: Usage of String NodeIDs

stephanhensel, 2017-05-30:
Hi,

I tried to use string NodeIDs in a node XML. I always got _None_ IDs in the generated file. This is because there is a little  - I think - copy paste mistake in the namespace compiler.
Please have a look at 
[https://github.com/open62541/open62541/blob/0.2/tools/pyUANamespace/open62541_MacroHelper.py#L320-L322](url)

There the NodeID identifier should be changed to _string_ and the string representation of the node should be used.
I think it should look like this:
```
elif node.id().s != None:
  code.append(node.getCodePrintableID() + "->nodeId.identifierType = UA_NODEIDTYPE_STRING;")
  code.append(node.getCodePrintableID() + "->nodeId.identifier.string = UA_STRING_ALLOC(\"" + str(node.id().s)+ "\");")
```

But there is also still the issue #717 open regarding the single and double quotes. This needs to be fixed to get string NodeIDs working. Currently I using a very dirty workaround in the _open62541_MacroHelper.py_ replacing the single quotes in the stored NodeID. Because my XML files uses single quotes which are stored within the NodeID object of the namespace compiler. For example: 

```
<UAObject BrowseName="1:Befuellen" NodeId="ns=1;s='TEST.XXX'">
```

is stored as `'TEST.XXX'`.

Best regards,
stephanhensel

jpfr, 2017-05-31:
@ichrispa Can you look into this?

ichrispa, 2017-06-12:
Hi @jpfr,

the issuer is setting opposite to my desk... I'll take care of the issue (origin of the problem is known).

regards,

ichrispa

Pro, 2017-08-13:
@ichrispa is this already fixed? Should we close the issue?

Pro, 2017-09-07:
Fixed in #739

closed 2017-09-07 by Pro